TriForceX/MiyooCFW #310: how change the reset button?

ahahahaah, 2022-02-19:
hello,
when i launch the game & watch emulator, the select button is supposed to display a menu, but this key is also set to leave the emulator so each time i want to use it, i leave the emulator and can't play.
who know how to fix the issue? thx

salvacam, 2022-02-19:
Have you tried the Retroarch core?
https://github.com/TriForceX/MiyooCFW/wiki/RetroArch-Setup

ahahahaah, 2022-02-19:
i'd prefer to use gmenu and launch everything with it, its more convenient like that

salvacam, 2022-02-19:
> i'd prefer to use gmenu and launch everything with it, its more convenient like that

I also. It is possible to integrate RA in gmenu2x, read the link that I have put.
You always have the option to compile the source code and change the buttons, I don't know where the standalone emulator source code is

ahahahaah, 2022-02-19:
i don't know what "compile the source" means^^ 
im not a developper nor a linux user, i just understand basic handling files.
but it's ok if it's a problem only for one emulator i can do without it :p
thx

Rezzy-dev, 2022-02-20:
The Select button on Miyoo is actually the ESC button -- which is why this emulators quits when you press it. Whoever successfully ported it didn't remap the default key bindings in the software to match the Miyoo layout (ESC to Right CTRL).

For Game & Watch the bundled emulator is not very good -- it doesn't scale the screen and only a few select horizontal layout G&W games are playable with it. RetroArch has a more up-to-date version of this emulator that supports scaling -- so you can play most if not all G&W games with it.

Rezzy-dev, 2022-02-26:
See here on how to set up Game & Watch through RetroArch: https://github.com/TriForceX/MiyooCFW/discussions/305

And the RetroArch setup documentation: https://github.com/TriForceX/MiyooCFW/wiki/RetroArch-Setup

ahahahaah, 2022-02-26:
thank you, finally if it is the only emulator that have a problem, i can do without it.  I will keep gmenu ^^
but indeed your post can interest other people

Rezzy-dev, 2022-02-26:
These Game & Watch games do work on the current emulator (in GMenuNX), @ahahahaah, without needing to set anything up, out of the box:

Donkey Angler.mgw
Egg.mgw
Explorers of Space.mgw
Fowling.mgw
Frog Boaster.mgw
Mickey Mouse.mgw
Parachute.mgw
Snoopy Tennis.mgw
Tom & Jerry Popper.mgw
Turtle Bridge.mgw

You still need to press the Select button instead of the Reset button to quit the emulator, but otherwise everything works fine, as expected, with them. ^_^

Rezzy-dev, 2022-02-26:
The original G&W (and Tiger Electronics, etc.) handheld devices were a bit like that, too. The controls were often strange (including their labelling) and required getting used to. Each device did its own thing, and there was no way of changing the controls.

(I remember playing them as a kid, and always needing to figure out the controls with each new device.)

But yeah, it would be nice if we could adjust it. The only problem is that even if you could adjust it, the current emulator for G&W doesn't support saving the configuration -- so you'd need to adjust it every time you play, which could get rather tedious.

Apaczer, 2023-03-11:
G&W standalone updated: <https://github.com/salvacam/gw_miyoo/releases/latest>

closed 2023-03-11 by Apaczer