TriForceX/MiyooCFW #314: Some Game Boy Advance games don't work properly with GPSP Rumble

Mamertoide, 2022-02-21:
Hi, I found a lot of crashing while playing some Game Boy Advance games, all them played with GPSP Rumble and using the GBA bios in my Pocket Go v1.

- Golden Sun: The game always crashed after defeat the boss Kraken.
- The Legend of Zelda - The Minish Cap: The game always crashes while playing the last dungeon.
- Mario Golf - Advance Tour: The game always crashes while you are creating a new character
- Mario Tennis - Power Tour: Same as Mario Golf
- Banjo-Kazooie - Grunty's Revenge: The game crashes after the title screen
- Castlevania - Harmony of Dissonance: The game crashes in the final battle against Dracula

I hope this will be fixed in the next update, thanks for your hard work.

Rezzy-dev, 2022-02-22:
* Batman Begins crashes at the fire scene on the first level.
* Teen Titans has graphical rendering issues (right at the start, in the intro and game menu).

jSTE0, 2022-02-23:
The upstream gpSP stopped development in 2009, it's unclear if any of the downstream forks are working on fixing the emulation core. See https://emulation.gametechwiki.com/index.php/GpSP for a list of forks and a compatibility list (with links to other compatibility lists). Some game entries mention that it used to work in a different version which is not an uncommon issue with non cycle-accurate emulators, such as gpSP (which trades accuracy for performance).

Rezzy-dev, 2022-02-23:
@jSTE0, are you aware of any more optimised forks of VirtualBoyAdvance-M or mGBA by any chance? Something that runs fast enough on ARM to play GBA games on (even if not quite as fast as gpSP)? Because at the moment gpSP seems to be the only option we have on the platform, and quite a lacking option to be honest -- it very much feels like unfinished work.

Apaczer, 2022-02-23:
https://github.com/rik-smeets/gpsp/releases

There was improvement done lately to stock gpSP emulator by @rik-smeets, who added autosave/load function, improved video filtering options and some others. You won't see any basic emulation improvement, but these are some features that community was really waiting for.

For best compatibility of that emulator I recommend to try it's RetroArch core on Miyoo. The development of it is very active and overheads compering to the standalone version are not so noticable (at least in my scenarios). You will find list of compatibility issues in RA below:
https://docs.libretro.com/library/gpsp/#compatibility

jSTE0, 2022-02-23:
I was about to post about the RetroArch gpSP core, it appears one developer has been working on the emulator core by improving its dynamic recompiler and porting it to other CPUs over the past 12 months. Some bugs have been fixed along the way so its game compatibility will be different.
I haven't gotten round to trying RetroArch on the V90 yet but will soon.

Rezzy-dev, 2022-02-23:
> There is a RetroArch mGBA core, though mGBA is supposed to be more accurate than gpSP so it will be slower.

Yep, mGBA is the complete opposite of gpSP in that it aims for perfect, cycle-accurate and authentic emulation, even at the cost of performance.

VBA-M (or its predecessor, VBA-Next) is a sort of compromise between the two -- it focuses on wide compatibility with all GBA games, but at the cost of emulation accuracy (but not speed so much).

I think I'll have a second look at @gameblabla's port of VBA-Next to ARM, and see if I can better optimise it:
https://github.com/Rezzy-dev/vba-next-portable

We really need an alternative to gpSP, too, on the platform. gpSP shouldn't be the only GBA emulator on Miyoo.

daFourch, 2022-07-29:
> https://github.com/rik-smeets/gpsp/releases
>
> There was improvement done lately to stock gpSP emulator by @rik-smeets, who added autosave/load function, improved video filtering options and some [...]

Nice thing if its development continues. At the moment, performances are not as good as on standalone version on Powkiddy V90

Apaczer, 2023-03-15:
Moving to discussion section & pinning for anyone feeling to add any "new" game to the [compatibility list](https://github.com/TriForceX/MiyooCFW/wiki/GPSP) of gpSP.