TriForceX/MiyooCFW #315: Flashing on Mac Instructions don't work

kerberw, 2022-02-23:
I tried following the Flashing on Mac Instructions.

Tried it 2 times, ended up with a card that just black screened on powkiddy v90.  The resulting card wouldn't even mount on the Mac either. Same card worked on PC with the same instructions.

Tried on MacBook Air M1 and MacBook Pro Intel, both running 12.2.1

Thanks!

Rezzy-dev, 2022-02-23:
Did you successfully unmount the card in step 3 before writing the disk image file to the card?

Make sure that neither the SD card nor the disk image are mounted when you are writing/flashing.

(In any case, if you have access to a Windows or Linux PC, it's a lot easier to flash the card there.)

kerberw, 2022-02-25:
Yes I did unmount the disk before. It showed as unmounted in disk utility.

Also, I do have windows access, although I use a Mac mainly. It works as described on Windows, was hoping to make you aware of the issues on Mac and I am available to troubleshoot to help fix the instructions for those who don't have access to windows or linux.

Rezzy-dev, 2022-02-25:
Glad to hear that, @kerberw, and I'm happy to work through this with you, and see if we can find out what the problem is. I suspect it has something to do with the changes Apple has been making to the way it handles non-Apple volumes lately. They seem to be targeting Windows technologies in particular (FAT32 is a Microsoft-created standard/format).

There is a bit of corporate warfare going on behind the scenes, and Apple hasn't been playing the game clean lately.

But these are just my suspicions, based on the evidence I've been presented with.

I don't have a Mac computer anymore, but I have a Linux laptop (practically also Unix-based), so together maybe we can figure what is going on with Macs.

Rezzy-dev, 2022-02-25:
The other thing I can think of -- although I doubt that is the issue -- is to make sure that you erase all volumes from the SD card before you unmount and flash it.

Rezzy-dev, 2022-02-25:
What happens if you add "bs=512" to the end of the "dd" command as you flash the card?

`sudo dd if="IMAGE.img" of="/dev/DISK" bs=512`

(Substitute IMAGE and DISK with the path to the disk image and the device name of the SD card you're flashing to, of course -- which you can double-check in Disk Utility.)

"bs" stands for block size. This will specifically tell dd to use the default block size for writing the image to the SD card. (In case Apple has changed something in how dd operates normally.)

kerberw, 2022-02-26:
> The other thing I can think of -- although I doubt that is the issue -- is to make sure that you erase all volumes from the SD card before you unmount and flash it.

So I am trying to erase the volume and reformat it to FAT in disk utility first, then the stock commands I will let you know as soon as that is done.

kerberw, 2022-02-26:
> What happens if you add "bs=512" to the end of the "dd" command as you flash the card?
>
> `sudo dd if="IMAGE.img" of="/dev/DISK" bs=512`
>
> (Substitute IMAGE and DISK with the path to the disk image and [...]

Trying this now. Will update when done.

Rezzy-dev, 2022-02-26:
When you check the flashed SD card in Disk Utility after you eject it and insert it again, what does Disk Utility show?

Also, can you paste here the results of 'diskutil list' in Terminal with the SD card inserted?

That may tell us what dd has actually done with the SD card, and may give us a clue as to why things are not working.

Rezzy-dev, 2022-02-26:
Can you paste me here all the steps you're taking and commands entering as you flash the SD card? I just want to make sure that you're following all the steps correctly.

It's very unusual how it's still not working, and especially how the card won't mount on the Mac after flashing.

Asppire, 2022-02-26:
> I tried following the Flashing on Mac Instructions.

See my post in the discussion forum:
[https://github.com/TriForceX/MiyooCFW/discussions/223#discussioncomment-2211901](url)
https://github.com/TriForceX/MiyooCFW/discussions/223#discussioncomment-2211901

Rezzy-dev, 2022-02-26:
> See my post in the discussion forum for an easier way:
> https://github.com/TriForceX/MiyooCFW/discussions/223#discussioncomment-2211901

Be VERY careful with Apple Pi Baker! It's not a supported method here, because ApplePi-Baker is a rather hacky homebrew application that can mess up your system with just a few clicks if you're not careful!

kerberw, 2022-02-27:
> When you check the flashed SD card in Disk Utility (after you eject it and insert it again), what volumes does Disk Utility show on the card?
> 
> Also, can you paste here the results of `diskutil list` in Terminal with the SD card inserted?
> 
> That may tell us what dd has actually done with the SD card, and may give us a clue as to why things are not working.

> > Update: That didn't help same issue, black screen no boot, disk was unmountable on Mac as well.
> 
> The SD card was unmountable on the Mac?? o.O

Yeah so the disk was totally unmountable on the Mac and PC says it needs to be initialized when plugged in.

kerberw, 2022-02-27:
> Can you paste me here all the steps you're taking and commands you're entering into Terminal as you flash the SD card? I just want to make sure that you're following all the steps correctly.
> 
> It's very unusual how it's still not working, and especially how the card won't mount on the Mac after flashing. I'm starting to wonder if it may be a hardware problem with your Mac (with the SD card reader/writer, or its driver).

diskutil unmountDisk "/dev/disk4s1"
sudo dd if="~/Downloads/Powkiddy-V90-Q90_CFW_v1.3.3_04_10_20.img" of=/dev/disk4s1

The output of this:
`7815168+0 records in
7815168+0 records out
4001366016 bytes transferred in 1603.007021 secs (2496163 bytes/sec)`

kerberw, 2022-02-27:
> Can you paste me here all the steps you're taking and commands you're entering into Terminal as you flash the SD card? I just want to make sure that you're following all the steps correctly.
>
> It's [...]

I tried on 2 different Macs, one an M1 and one an Intel. Same problem on both! Running latest public release: 12.2.1
Same card, same download, same reader all work on PC.

jSTE0, 2022-02-27:
> Flash w/: sudo dd if="~/Downloads/Powkiddy-V90-Q90_CFW_v1.3.3_04_10_20.img" of=/dev/disk4s1

I think that's the wrong device. I don't have a Mac but that looks like disk 4 partition 1 to me which means the image ended up inside the FAT partition, which is why it ends up unmountable in any OS.
Based on instructions from https://osxdaily.com/2018/04/18/write-image-file-sd-card-dd-command-line/ I think it should be:
```
sudo dd if="~/Downloads/Powkiddy-V90-Q90_CFW_v1.3.3_04_10_20.img" of=/dev/rdisk4
```
so that the entire drive is overwritten as the image includes its own partition table and the separate partitions for boot, root and main.

kerberw, 2022-02-28:
PEBKAC? So in my case it would be sudo dd if="~/Downloads/Powkiddy-V90-Q90_CFW_v1.3.3_04_10_20.img" of=/dev/disk4 instead of disk4s1 ?

Thanks for seeing this, I used the identifier in disk utility once unmounted.

jSTE0, 2022-02-28:
No, `of=/dev/rdisk4`, note the `r` for raw disk.
I wouldn't call it PEBKAC, the instructions in the wiki are, in hindsight, a little vague if you're not familiar with partitions vs whole disks. It wasn't until I saw `of=/dev/disk4s1` that I thought "that doesn't look right". The equivalent in Linux would be `/dev/sdc4` for the 4th partition of the 3rd SCSI/SATA/PATA/USB disk compared to `/dev/sdc` for the whole of the 3rd disk itself.
Do double-check that the rdiskNUMBER is correct though.

Rezzy-dev, 2022-02-28:
Correct. Either "disk#" or "rdisk#" are fine, but NOT "disk#s#", seeing as the "s" number stands for "sector" number (meaning the first physical partition/volume on the disk). It's a strange labelling system that Mac OS uses, but that's how it works.

You need to write/flash to the disk directly, NOT to a sector/partition in the disk.

I will update the documentation to better reflect/detail this unusual labelling system in Mac OS.

As for the unusual "r", here is the explanation behind it:

> Using Mac OS X
> 
> Note: In OS X each disk may have two path references in /dev: /dev/disk# is a buffered device, which means any data being sent undergoes extra processing. /dev/rdisk# is a raw path, which is much faster, and perfectly OK when using the dd program. On a Class 4 SD card the difference was around 20 times faster using the rdisk path.

So either "disk#" or "rdisk#" are fine using dd on Mac OS, but "rdisk#" tends to work faster.

Rezzy-dev, 2022-02-28:
Documentation updated:

> Note: The disk# does **not** include the s# following it! The s# stands for sector (partition/volume) number in Mac OS. You need to write to the disk directly, not to a partition on the disk. If you write to a partition on the disk, you will need to re-flash, as the flashed SD card will not work/mount.

https://github.com/TriForceX/MiyooCFW/wiki/Custom-Firmware-Setup

kerberw, 2022-03-01:
I tried this with the updated help, and yea! It worked! I will close this, thanks for helping me resolve it!

closed 2022-03-01 by kerberw

Rezzy-dev, 2022-03-01:
Thanks to you, @kerberw, for your brave openness, team spirit, and persistence, and to @jSTE0, who chipped in with a bit of very useful insight even I was not aware of! :+1: ^_^ <3

Glad to know that the Mac process/documentation will be much clearer to follow from now on. This was a great team effort! \*^_^\* <3