TriForceX/MiyooCFW #316: [Bittboy v3.5]

nisrulz, 2022-02-24:
I have a BittBoy v3.5 (s/n ends with 005)

[image]

I have followed the English guide as mentioned [here](https://github.com/TriForceX/MiyooCFW/wiki/Custom-Firmware-Setup#1-materials-1)

[image]

I use 
- macoS.
- Software used to flash: ApplePieBaker
- Software used to expand the `main` partition: iPartition

After I flash with ApplePieBaker, I end up with correct partitions i.e `boot` and `main`. I transfer some ROMs and put it in the BittBoy, but there is only black screen. 

I have tried the brightness check, as well as went ahead and edited the `.backlight.conf` file and set the value to `8`.

I also found this forum post suggesting to replace the kernel of v3.5 with v3 firmware [here](https://boards.dingoonity.org/other-game-systems/bittboy-3-5-randomly-booting-to-black-screen-fix-(bittboy-3-5-only)/). The file in link doesn't exists anymore.

At this point I am just lost how to get my BittBoy to even work. If anyone has 
1. Kernel image of v3 that I can attempt that trick also
2. Original firmware image

I would be happy to use the very old version of Bittboy since it was atleast functional for me.

TriForceX, 2022-02-24:
Did you get the "Miyoo" logo at boot? 
Can you try to flash only (without expand)?

nisrulz, 2022-02-24:
Tried just flashing just now.

- No logo shows up
- I hear an audible bell sound
- Screen stays black
- Re-checked brightness again via buttons and also the config file. Looks find to me.

TriForceX, 2022-02-24:
That should be an issue with the screen driver... NOT all the consoles has the specific version parts in the hardware (for example there are some bittboy v2 with a v3 screen, or a pocketgo with a v2 screen...).

Im not sure if this is the problem here, but can you try to use the screendrivers from the other versions in your 3.5 install? the driver is "r61520fb.ko" and is placed on Main partition > Kernel folder

nisrulz, 2022-02-24:
So are you suggesting I copy paste r61520fb.ko file from other Firmwares i.e v2/x3 and see if it works with that?

I do not need to re-flash I am guessing.

nisrulz, 2022-02-24:
Is there a way to check which screen version I have as hardware part? I would like to narrow it down to what I am looking for.

TriForceX, 2022-02-24:
> So are you suggesting I copy paste r61520fb.ko file from other Firmwares i.e v2/v3 and see if it works with that?
> 
> I do not need to re-flash I am guessing.

Yes.

> [I found this link with older Firmwares](https://rs97.bitgala.xyz/Miyoo%20Devices/Bittboy%20cfw/). Maybe I should just see copy pasting r61520fb.ko file from each and see if which one works.

No, use the same drivers from 1.3.3, previous version may not work, with the current 1.3.3.
Anyways i have all the older firmware versions, i just removed the old ones from dropbox/gdrive to keep the latest 1.3.3.

> Is there a way to check which screen version I have as hardware part? I would like to narrow it down to what I am looking for.

No, sadly the hardware in this kind of consoles is a mess, we tried to checked them before and the system gave us different results each time.

nisrulz, 2022-02-24:
So I downloaded all 3 firmware
- v2-2.5
- v3
- v3.5

Installed v3.5 and replaced r61520fb.ko file from v2 and then v3. In both cases its the same case. The SD Card has also been replaced so the possibility of it being corrupted is not there.

You said:
> Anyways i have all the older firmware versions, i just removed the old ones from dropbox/gdrive to keep the latest 1.3.3.

Could send me the original firmware for Bittboy v3.5? As mentioned earlier I am not concerned about the new fixes. I would like my device to work as it did earlier.

Rezzy-dev, 2022-02-24:
Note that since you're using a Mac, this may be a latest MacOS hardware/software related issue, too: https://github.com/TriForceX/MiyooCFW/issues/315

If you have access to a Windows PC (or Windows partition on your Mac), have you tried following the Windows flashing instructions?

https://github.com/TriForceX/MiyooCFW/wiki/Custom-Firmware-Setup

Mac seems to be doing some rather unorthodox things lately with non-Mac, and FAT32 volumes...

Also, are you using the SD card that came with your device (from China), or a proper SD card you've purchased separately (in your country)? The SD cards that are supplied with the devices tend to be faulty/hacked (to leave a bigger profit margin), and this could be the case here, too.

nisrulz, 2022-02-24:
I'll find a way to run Windows VM on my mac or use a Linux machine.

SD Card I bought a new one from Amazon and is from SanDisk, so pretty sure it is fine.

I'll keep this thread updated if I can get it to work.

Rezzy-dev, 2022-02-25:
If that doesn't work, you could also try using the PocketGo or PowKiddy version of the firmware for flashing. It's possible that the manufacturer may have used hardware components from these later devices in your BittBoy to save on manufacturing costs (because they had more of these in stock). So it's possible that the screen may work with one of these other firmware/drivers.

You'll need to experiment a bit to see what works. I bet that the components used are not brand new kinds of components, but rather an unusual mix-and-match of existing ones in the current range.

Rezzy-dev, 2022-03-01:
@nisrulz, the Mac flashing documentation/guide has been updated. There's no hardware problem with the Mac flashing (the process still works), but you need to make sure **not** to include the sector number (s#) when pointing to the SD card disk# to flash to. If you include the s#, too, the resulting flashed SD card won't mount and the firmware won't start (as you will have written partitions inside a sector/partition -- which is a no-no in terms of disk formatting).

nisrulz, 2022-03-01:
Thank you for the update @Rezzy-dev . 

I have been using ApplePieBaker, so I wasn't specifying any sector number manually. Maybe I should do it the way the docs in this repo's wiki says to do. I'll attempt that also. Thank you for the headsup. Would be really nice if my dead device starts working, I was very fond of it :)

Rezzy-dev, 2022-03-01:
Yep, give the official method a go. It's been tested multiple times now on various Macs, and it delivers safely/reliably. Try it without expanding the FAT32 partition ('main') first, and see if the firmware boots and the display shows.

jSTE0, 2022-03-02:
I think the flashing is being done correctly already as the correct partitions are present and mountable, and the device boots (startup beep is heard).
It looks like there might be a fix in a newer kernel from MiyooCFW/kernel#9, or at least some way of trying different panel types. I guess it's either wait for the new CFW release or hope someone has a copy of the original firmware, perhaps try the Discord channel?

Rezzy-dev, 2022-03-02:
I'm sure bringing it up on the Discord channel can help.

And yeah, in your individual case it looks like the firmware is working, it's just that your device has a different combination of physical components, and the BittBoy firmware's video driver doesn't work with your device. Have you tried the other versions of the firmware?

nisrulz, 2022-03-02:
- I have tried to join Discord, but it doesn't let me join. This is why I created the issue in the repo. 
  [image]
- Attempting PocketGo/PowKiddy kernel is on my list. It is just that I have tried so many things that I mostly confused what works or what I have already attempted 👀 
- I plan on doing the below in order:
 1. Use official mac flashing steps with the latest firmware for BittBoy 3.5
 2. Replace the kernel with that of PocketGo/PowKiddy
 3. If I can get into Discord somehow, then ask for an original backup of firmware. This was a failure on my part where I never created a backup before attempting the latest firmware (or atleast keeping the kernel file). Well, good learning still for future.
 4. As my last resort I will open up the device and try to lookup hardware details and try to compile the firmware + kernel myself. I am not letting this device get thrown out until I have put everything at it 😅
 
Appreciate all the suggestions/recommendations. I atleast have some pointers on what could be happening here.

Rezzy-dev, 2022-03-02:
You've left out one very important step (or a few):

- Use official mac flashing steps with the latest firmware for PocketGo and PowKiddy Q90.

You'll need to try to flash these later versions of the firmware, as the hardware in your BittBoy may be made of PocketGo components, for example. (This is what I suspect to the be the most likely culprit.)

jSTE0, 2022-03-04:
@nisrulz I noticed in the beta testing section there are replacement screen drivers to reduce screen tearing. Screen_Tearing_3 says it works on BittBoy 3.5. It's meant for an older version of the CFW (one that can no longer be downloaded). You could try that and see if it brings your device back to life.

nisrulz, 2022-03-04:
@jSTE0 Can you add a link to where you read it?

jSTE0, 2022-03-04:
@nisrulz Go to the Beta Testing folder https://drive.google.com/drive/folders/1AT3LVlJPCZf_csTh8qdgL0Bp1MlZQONV and it's `Screen_Tearing_3.txt`. It may or may not work with the kernel on the newer CFW, just thought it'd be worth a try.

nisrulz, 2022-03-04:
Just attempted. Doesn't work 👀

jSTE0, 2022-03-04:
Oh well. I guess you're left waiting for someone to upload the original firmware or compiling the newer kernel module.

nisrulz, 2022-03-04:
I see in Gdrive [this](https://drive.google.com/drive/folders/1PAYQC8loV2114ips-0wjMxzbXa4VdLp5) Its called Stock Firmware. How can I use this?

jSTE0, 2022-03-04:
That's only for BittBoy v1 which boots from SPI flash on the motherboard (if you look at the flashing instructions for that model you'll see how much more difficult it is). BittBoy v2 and newer boot from microSD.

nisrulz, 2022-03-04:
Update on what I have tried

None has worked for me yet.
- Use official mac flashing steps with the latest firmware for BittBoy 3.5
- Use official mac flashing steps with the latest firmware for PocketGo/PowKiddy.
- Replace the [kernel](https://github.com/MiyooCFW/kernel) with that of PocketGo/PowKiddy into Firmware of BittBoy 3.5.
- Replacing Beta Testing screen_Tearing_3.zip kernel with one in Firmware of BittBoy 3.5.

I am now looking up 
- how to compile the Kernel with the fix mentioned above by @jSTE0 and drop it in my firmware.

P.S: Is it possible for me to get an invite for Discord? I think it would be helpful for me to discuss with others. The current link in repo, don't let me in.

Rezzy-dev, 2022-03-05:
o.O How curious! So the hardware components inside your BittBoy are something new after all? :O

TriForceX, 2022-03-05:
@nisrulz i cant send you the backup image from 3.5 due contains illegal stuff (roms) so ive extracted some of the main parts so we can figure out about our issue:

bittboy v3 backup:
https://www.dropbox.com/sh/jfbuj9w1nwjis6i/AAB_TwMQc-zwWhcI4OATbx1Ka?dl=0

bittboy v3.5 backup:
https://www.dropbox.com/sh/88xggiviqoadjci/AADBjKnMOX6i70t8lqSHfhkOa?dl=0

btw this is the discord link you tried? https://discord.gznetwork.com

nisrulz, 2022-03-07:
@Rezzy-dev Unfortunately the seller doesn't respond, which is why I am stuck. I tried contacting them and no reply for months. 

If it was h/w issue I would have left it when the seller didn't follow through. Ofcourse I can buy a new one, but the fact that this is a software issue is why I am even pursuing this and totally believe I can find a way to fix it.

@TriForceX I'll take a look at the links and files. Thank you.

Also the Discord link still doesn't let me join:

[image]

[image]

I believe your generated link has limit on how many users can use it and now it has expired after multiple people used it and joined. You probably need to generate a new one.

Rezzy-dev, 2022-03-07:
> If it was h/w issue I would have left it when the seller didn't follow through. Ofcourse I can buy a new one, but the fact that this is a software issue is why I am even pursuing this and totally believe I can find a way to fix it.

Well, technically it **isn't** a software issue. That's just it. It's unsupported hardware. It can be fixed in the software (support can be added for it), but it's not a software issue.

But I understand why you're trying -- because it worked when you got it, and there's a way to make it work again (even if it takes quite a bit of tinkering/work).

Although if I was walking in your shoes, I would have probably returned the unit, gotten a full refund, and purchased a new unit from a different seller. After all, you didn't get what you paid for, what you agreed to buy -- you got something different, and it looks like something very unique. It's your personal choice, though, what to do with it.

And if you're firm enough in your resolve and empassioned/committed to finding a solution, I'm confident you will. You'll find support here. :)

TriForceX, 2022-03-07:
I agree with @Rezzy-dev, sadly this happens when manufacturers doesnt support developers, usually (in most of cases) they use and modify the cfw to their needs and never sent back any changes to the main project. The other bad thing is the last years they started to reuse parts/components from other versions of the console (or maybe they make new ones). At this moment there are other 2 guys with the same problem as you... so we have to figure out what the manufacturer changed.

We can support you, even if we have to make a custom version of the modules/driver to have your console working :) Tell us if you have some progress with the files i sent you.

Btw about the discord link idk why isnt working, ive checked and is set for unlimited usage and users... you can use this one for the moment https://discord.gg/p2jgjUG4

symbios24, 2022-04-10:
Just find a windows pc in 5 mins you be ready
Mac are known for huge incompatibilities

Rezzy-dev, 2022-04-12:
> Just find a windows pc in 5 mins you be ready Mac are known for huge incompatibilities

That's personally biased, and technically not correct. Windows has just as many incompatibilities with Linux as Mac has. In fact Mac is a little closer to Linux, as they are both Unix-based-design software and share some core system software.

The reason that Windows appears to be more compatible is simply that is has more graphical UI software written for the platform, seeing as Windows is the most used operating system out there.

But popularity does not equal quality, or greater compatibility, necessarily.

TriForceX, 2023-04-10:
Closed due inactivity.

closed 2023-04-10 by TriForceX