TriForceX/MiyooCFW #321: snes9x4d speedhacks file snesadvance.dat in wrong directory

jSTE0, 2022-03-04:
There are 2 copies of `snesadvance.dat`  shipped shipped with CFW 1.3.3, one in `.snes9x4d` and another in `emus/snes9x4d` but neither is where the emulator expects it to be.
The version of snes9x4d shipped is v2020.06.10. In m45t3r/snes9x4d@0c86522f for the v2020.06.03 release, the configuration directory changed to `.snes9x4d-ng` so that's where `snesadvance.dat` should be placed.
Once moved, the message `Found speedhacks, applying...` appears on startup.
This issue should be fixed for the next CFW release.

jSTE0, 2022-05-12:
From #380, perhaps it should be left out as it can cause game breakages, maybe only on specific emulators or emulator versions.

Rezzy-dev, 2022-05-13:
No, it greatly boosts performance for most games -- it should definitely be kept as an essential companion to Snes9x4D! :D

What I would suggest instead is for there to be a user interface setting in the emulator to turn the speedhack on or off on a per-game basis (saved in the game's config file). On by default, and can be switched off by the user for a game if the speedhack happens to break compatibility with that game.

That way we get the best of both worlds -- both wider compatibility and faster performance.

filipeaguiar, 2022-07-22:
Is there any problem on putting a copy of the file on the `.snes9x4d-ng` directory, instead of moving it?

Rezzy-dev, 2022-07-22:
Of course not. If you wish to keep a copy of it in the old directory, you can. There's not much practical use for keeping it in the old directory, to my best knowledge, but you can have it in both locations if you want to. So copying is fine, too -- you don't _have to_ move it.

Apaczer, 2023-03-15:
The focus will be only on releasing bare stable CFW img from now on and every other app beside essential ones (e.g. gmenu2x, st-sdl) will be threated as 3-party thus provided separately - installing them is only up to individual user.
Moving to discussion if anyone want to give more feedback on issue or use developer's main repo of snes9x4d: https://github.com/m45t3r/snes9x4d/issues