TriForceX/MiyooCFW #330: Mplayer force closed

DoctorOey, 2022-03-20:
I have problem with mplayer in miyoo 1.3.3 powkiddy v90.
When im use stock fw mplayer is works well.
But after update to miyoo 1.3.3 the mplayer doesnt open/force close. Any suggestion??

Rezzy-dev, 2022-03-20:
Yep, it's a known bug in the current version of MPlayer on Miyoo. When trying to resume a saved playback progress/state, it crashes. Extract and put this in "/apps/mplayer/" (replacing the file there):

[run.zip](https://github.com/TriForceX/MiyooCFW/files/8310799/run.zip)

It amends the run script so that .RESUME files are never left behind on exit. MPlayer won't remember your playback progress/state anymore, but it won't crash either, and you'll be able to use it.

DoctorOey, 2022-03-30:
Sorry , i wanna ask again.
What a support video resolution/bitrate/fps/extension format in mplayer? This cpu sheet said support full hd playback?

Rezzy-dev, 2022-03-30:
Miyoo devices do not have a graphics card. You won't really be able to play HD video on them. In fact I would expect all but the simplest to play videos to stutter.

The marketing for it mentions HD video support, but it's not realistic:
https://www.allwinnertech.com/uploads/pdf/20181218155101eb.pdf

We're working with a 700 MHz ARM processor, and 32 MB of RAM here. And that's all of it! So don't expect much in terms of video playback. For HD video you really need a graphics card.

jSTE0, 2022-03-30:
The HD video decoding support (though only 720p30, the F1C200s supports 1080p30) is available in  refers to the Video Engine part of the SoC. A [kernel driver](https://github.com/aodzip/cedar) and [userspace library](https://github.com/aodzip/libcedarc) do exist but the latter exposes an OpenMAX interface which isn't supported by MPlayer. A different player could be used like OMXPlayer but all of that needs porting and I'm not aware of any such efforts. That means all decoding, colourspace conversion and scaling is done by the CPU.
As for video format limits, etc. I'd suggest a video that's close to the native 320x240 resolution of the screen and I suspect that's about as much as the CPU can decode and scale in real-time. A higher resolution video like 480p would end up needing even more CPU to downscale it to the screen's native resolution.

DoctorOey, 2022-03-31:
> The HD video decoding support (though only 720p30, the F1C200s supports 1080p30) refers to the Video Engine part of the SoC. A [kernel driver](https://github.com/aodzip/cedar) and [userspace library](https://github.com/aodzip/libcedarc) do exist but the latter exposes an OpenMAX interface which isn't supported by MPlayer. A different player could be used like OMXPlayer but all of that needs porting and I'm not aware of any such efforts. That means all decoding, colourspace conversion and scaling is done by the CPU. As for video format limits, etc. I'd suggest a video that's close to the native 320x240 resolution of the screen and I suspect that's about as much as the CPU can decode and scale in real-time. A higher resolution video like 480p would end up needing even more CPU to downscale it to the screen's native resolution.

Yes, im already try any resolution. In fact only Mpeg4/320x240p/1024kbps/20fps only the video running smoothly.

Thanks for All reply.

closed 2022-03-31 by DoctorOey