TriForceX/MiyooCFW #331: Pcsx reArmed Stability

DoctorOey, 2022-03-23:
I have problem with the latest pcsx oct 2019 and config part 2 version.
When playing Cyborg kurochan / metal slug x, it not produce sound. When i change the bios from hle to scph1001, sound present, but the controller can enter the game. Any suggestion which pcsx version is stable.?

Nb : all rom i converted to pbp files. And i dont know the version (pal/ntsc).

Rezzy-dev, 2022-03-23:
All currently released mainstream versions of PCSX ReARMed are stable, but you may need to tinker around with the settings a bit to get a game to work properly.

There are community-made config (CFG) file bundles you can download where people have set up PCSX ReARMed for certain games. That can make things a bit easier to start with.

There are some in the Beta Testing archive: https://github.com/TriForceX/MiyooCFW#beta-testing

You could try using the "psxonpsp660" BIOS, too. It works better with some games.

Note that Metal Slug X is also available for NeoGeo, and it works much better there. (You'll just need to copy the "neogeo.zip" ROM dump into the same ROM folder as the game.)

DoctorOey, 2022-03-23:
Thank you for responding.
I have fix it by change bios.
So some japanese iso need scph 1000 works well. 1001 not working.
Yeah i will try that bios.

closed 2022-03-23 by DoctorOey

Rezzy-dev, 2022-03-23:
Yep, the numbers correspond to region. So most likely 1001 is U.S., and 1000 is Japanese.

But as I said, you'll get much better performance with Metal Slug X on GnGeo. Metal Slug X was a game originally released on the Neo Geo, and then ported to the PlayStation.

DoctorOey, 2022-03-24:
Yes, i have been play it. Very smooth and zero lag. 
Does miyoo supoort psx4all? Like in anbernic series.?

Rezzy-dev, 2022-03-24:
I don't know of any PSX4All build for Miyoo yet, no. *Shakes head.* Perhaps in the future.