TriForceX/MiyooCFW #332: v90 image quality

itchyd, 2022-03-27:
The anbernic 280v is the same resolution as the v90, the screen is a similar size and yet the quality of text on both devices is waaaaaaay different.  The 280v is very legible while the v90 is kind of a mess.  What is going on here?

Rezzy-dev, 2022-03-27:
A mess? What do you mean? :open_mouth: 

Is it using a different font, maybe? You have rougher fonts and smoother fonts, and you can change the font in the skin settings.

TriForceX, 2022-04-01:
screen quality/type is different
hardware/firmware/emus/apps are different so screen scaling, filters, etc... are different

If you have a defailed example of what you want to know about please post a picture, but mostly emulators are non pixel-perfect so it depends how the emulator were made and how it deals with the screen scaling, also due the hardware is different some display features may not be available on the v90, so in general terms is not a good way to compare them... is like comparing gb versus gbc