TriForceX/MiyooCFW #333: Neo geo powkiddy v90 miyoo problem

DoctorOey, 2022-03-28:
Hi bro
I have issue with metal slug 3 and other game in neogeo new version in miyoo cfw.
The main problem is decrypted time loading is very loooooong and not run. 
When i use final burn alpha its working but the fps very poor.

Its a bug? Or what in neogeo? Have u try metal slug 3 in neogeo is stuck in decrypted loading.

Apaczer, 2022-03-28:
This is due to low RAM on these devices and you can load only ROMset zipped not bigger than 40mb with stock neogeo emulator.

However there has been some workdone lately adressing that issue and MS3 should be playable in newer revision. I'll post back here when I find working code with instructions.

Rezzy-dev, 2022-03-29:
Metal Slug X is the last version of Metal Slug that will work with the current release of GnGeo.

Games released past the first half of the year 1999 will not work. Metal Slug 3 was released in 2000: https://en.wikipedia.org/wiki/List_of_Neo_Geo_games

Rezzy-dev, 2022-03-29:
> This is due to low RAM on these devices and you can load only ROMset zipped not bigger than 40mb with stock GnGeo emulator.

The King of Fighters '98 is 41.1 MB zipped, and still loads and play fine. Not saying that what you said isn't the case, but if so, the limit is not 40 MB.

I always suspected it was due to the hardware change in 2000s Neo Geo game systems. After 1999, SNK upgraded their system(s) and made some big modifications to how the games work.

Apaczer, 2022-03-29:
Sorry, should be more precise it surely is not 40MB exactly, but anything larger thatn KoF'98 will be unplayable.

Rezzy-dev, 2022-03-29:
It's interesting, because the Miyoo devices have 32 MB of RAM. It's a techical question, but I wonder why 41 MB is the limit. That's 9 MB over the available memory. Has anyone tested the same build/version of GnGeo on a device with more RAM, like a PC?

I suspect it may not be the size of the RAM that is the reason for the failure to load the ROMs. Post-1999 ROMs are treated by GnGeo as if they were encrypted -- which suggests to me that GnGeo is not familiar with their formatting, and mistakes them for encrypted archives (attempting to decrypt them and obviously not succeeding, hanging in the process).

I think it may have more to do with the change in format (different hardware) in the ROMs themselves than the physical size of the zipped ROM.

Rezzy-dev, 2022-03-29:
Hmm... I'm pretty sure I read something about a change in the MVS boards/hardware in the early 2000s, but this would be a more likely culprit, and aligns with what we're both talking about:

https://en.wikipedia.org/wiki/Neo_Geo_(system)#ROM_sizes_and_startup_screens

Later in the life cycle of the system, at the beginning of the new millenium, SNK started increasing the amount of memory the games/ROMs used.

330 megabits is 41.25 MB. 716 megabits is a whopping 89.5 MB!

During the 330 megabit cartridge limit, most games consumed less than 41.25 MB of RAM memory -- bringing us into the safe territory of the 32 MB that Miyoo devices have. Once the available RAM in the cartridges was increased to 716 megabits, games/ROMs started using more than 32 MB of RAM. And this transition happened around the turn of the millenium, sometime in the second half of 1999 -- most likely starting with SNK's flagship title, The King of Fighters '99.

So it's not the size of the zipped ROM file, but the hardware change in the cartridges, the boosted RAM limit/requirement, that is causing the loading problem with post-1999 games in the current release of GnGeo on Miyoo devices. The games use more RAM than what the system has available.

Rezzy-dev, 2022-03-29:
SNK actually introduced the 716 megabit cartridge limit in 1998, but it looks like games didn't start going over the 32 MB RAM limit until the second half of 1999.

These are 716 megabit cartridge Neo Geo games:
https://www.giantbomb.com/giga-power/3015-3934/games/

(The 1998 titles work in the current build of GnGeo -- everything up to Metal Slug X, released in the first half of 1999.)

DoctorOey, 2022-03-29:
> It's interesting, because the Miyoo devices have 32 MB of RAM. It's a techical question, but I wonder why 41 MB is the limit. That's 9 MB over the available memory. Has anyone tested the same [...]

If metal slug 3 have over 40mb cant running in 32mb of ram, but why Ps1 over 700mb can running in this device? 😆. The problem is not in the Ram. The problem is in the emulator it self.*my opinion.

Rezzy-dev, 2022-03-29:
I remember the later released Neo Geo CD had this issue as well, actually. It had very long loading times, because it had a slow CD drive, and the games were essentially the same code as for the cartridges -- which meant that everything needed to be loaded into memory before the game could play.

Neo Geo games were designed/coded with dedicated and instantly accessible RAM in mind.

DoctorOey, 2022-03-30:
Any Someone have to try neogeo in anbernic series? How about that? Is running smoothly? Like RG 351MP/V series. It has 1gb ram and quadcore cpu. If the problem is same like in the powkiddy, i think it failure of development of that emulator (need some fix).

Rezzy-dev, 2022-03-30:
> Any Someone have to try neogeo in anbernic series? How about that? Is running smoothly? Like RG 351MP/V series. It has 1gb ram and quadcore cpu. If the problem is same like in the powkiddy, i think it failure of development of that emulator (need some fix).

Or the GnGeo emulator on Miyoo simply supports only 90s' Neo Geo games currently -- everything up to The King of Fighters '99, the second half of the year 1999.

This is true whether it's an insufficient RAM issue or an emulation support issue -- all the same.

Hopefully post-1999 game support will be added in the future.

closed 2022-03-30 by DoctorOey

Apaczer, 2022-04-25:
see https://github.com/TriForceX/MiyooCFW/discussions/369