TriForceX/MiyooCFW #356: Screen Blurred

DoctorOey, 2022-04-10:
Hi there, i got a problem, when trying play a playstation with rom digimon world 2003. The problem is when im fusion the monster, the screen got bluureddd.
Its software isssue? Emulator issue? Or hardware issue?. After fusion the monster the screen is back, and shadow of screen back to normal.
It happen again too, after startup in miyoo menu screen. Take 2-5 second then back to normal. Any suggestion?.

DoctorOey, 2022-04-10:
[image]

DoctorOey, 2022-04-10:
After im review,,this happen when the battery indicator is red. I think the cpu make the screen blurred to minimize power consumption for 3d graphic (fusion the monster have 3d animated).

TriForceX, 2022-04-11:
Can you attach photos?

DoctorOey, 2022-04-11:
[image]

Rezzy-dev, 2022-04-12:
How low are you letting the battery run before recharging? Make sure that you don't wait till the battery is completely depleted, as that will reduce the battery's lifetime. When the system is critically low on power, some hardware functions will be cut to preserve power and keep the system running -- this is normal in all mobile computers/devices.

Shortly after you get the first power warning, when the power indicator goes red in the main menu, you should power off your device and put it on charging.

DoctorOey, 2022-04-12:
> How low are you letting the battery run before recharging? Make sure that you don't wait till the battery is completely depleted, as that will reduce the battery's lifetime. When the system is [...]

Every 50% or 2 bar battery i always charging. 
But when im use original firmware its not happened.
It happen when use miyoo. 
This morning it happen again. Check this photos

DoctorOey, 2022-04-12:
[image]

I got this screenshot from fb, he has problem while in miyoo too. He use 3 different sd card use miyoo. Same problem. I dont know what the bug is..

Rezzy-dev, 2022-04-12:
You need to cleanly shut it down (not suddenly), yes, but that shouldn't be causing this. I think this may actually be a fault with your particular device/hardware, with your PowKiddy V90.

Does it happen when using different emulators, too?

Rezzy-dev, 2022-04-12:
> It happen again too, after startup in miyoo menu screen. Take 2-5 second then back to normal.
> Or hardware issue?

Definitely a hardware issue. It's an issue with the screen or display card. If it persists after you shut down the system and turn off the console, if you still see it on next boot, then it's most definitely a hardware issue, not software.

If you can get a refund on the device, I would. PowKiddy devices get made in all sorts of factories, and some of the factories produce sub-par products. Try buying the device from a different seller.

DoctorOey, 2022-04-12:
> Ah, now I understand what you mean. The screen refresh gets stuck, right? And the screen starts to gradually blur more and more, leaving trails?

Yes...it happen in miyooo, original firmware is ok

DoctorOey, 2022-04-12:
> You need to cleanly shut it down (not suddenly), yes, but that shouldn't be causing this. I think this may actually be a fault with your particular device/hardware, with your PowKiddy V90.
> 
> Does it happen when using different emulators, too?

Yes, it happened too, while in the menu Gmenu, emulator, sometimes and back to normal

DoctorOey, 2022-04-12:
Okay its the rootcause.
1. After flash to miyoo do not copy your old data (may corrupted before)
2. When u get suddenly shutdown, the card need reflash.
3. Make sure your sd card is okay
4. Max memory is 32GB for stability. SdXC may need extra voltage that make your powkiddy system unstable like screen tearing/gradation.

The key is do not copy your old data with history already sudden off/shutdown (it may corrupted).

Im use original firmware the problem is not found.

closed 2022-04-13 by DoctorOey