TriForceX/MiyooCFW #363: Add vibration motor to Powkiddy V90

360fun, 2022-04-15:
Hi and thank you for the amazing work behind this firmware! 🤖 

After I bought the Powkiddy V90 I discovered (https://youtu.be/nWTRpo2fzJI) that some consoles have a vibration motor and others don't (like mine 😢 ). I was wondering if it can be soldered to the motherboard since the software (and hopefully the circuit lines) are already there. I don't know much about circuits so it's just a guess/hope. 🤞

I couldn't find images about internals with the motor to compare, but in [these images](https://www.reddit.com/r/SBCGaming/comments/mx5wno/didnt_see_a_teardown_of_the_powkiddy_v90_anywhere/) that white circle with 2 nearby soldering points looks like the perfect spot: 
[image]

Internals from BittBoy V3.5:
[image]

Any expert could confirm/deny this guess?

TriForceX, 2022-04-18:
Here you have a pic from a v90 teardown
[image]
more pics here https://steward-fu.github.io/website/handheld/v90/teardown.htm

360fun, 2022-04-18:
Looks like I was right! 🥳 I will check with a tester if I get voltage when vibration is activated, thanks!