TriForceX/MiyooCFW #367: gbsp crashes

FloofTheWolf, 2022-04-20:
So have been using this CMF for a good while, and the people who make this all possible look like super nice people, but the GBA emulator crashes sometimes and that leaves me no choice but to turn it off with the power button beacuse no other buttons work.

So if anyone cloud help I would be more then thankful.

Rezzy-dev, 2022-04-20:
If an emulator/software freezes, hold the Reset (Minus) button and press Select. It will force the emulator/software to quit and shut down the console neatly. Then you can safely flip the switch, once the screen turns off, and turn the device on again.

It has very fast performance, but gPSP's game support is patchy at best, unfortunately. It will crash with some games and, evidently, freeze with others. There's no other GBA emulator for the Miyoo CFW at the moment, but there are some attempts both to further develop gPSP and to port new GBA emulators to the platform.

There are also attempts to document gPSP's game support.

FloofTheWolf, 2022-04-21:
> 

Oh! I didn't know it was still possible to shut it down! Well thanks for this info, i was just very worried about my stuff getting corrupted beacuse of this! Big thanks!

jSTE0, 2022-04-21:
RetroArch has gpSP, which is newer than the standalone version, and mGBA, which is quite slow for GBA emulation, cores. Whatever you tried to run might work on those.

Apaczer, 2022-04-21:
Maybe the compatibility list should be updated: https://github.com/TriForceX/MiyooCFW/wiki/GPSP ? Several people already reported other games having issue on standalone (some fixed in RA - e.g. text in Rayman 3).

@Rezzy-dev could you have a look into it, IIRC you had some problems with gpSP as well?

Rezzy-dev, 2022-04-22:
Thanks @Apaczer! I've updated the list and added further means of contact/reporting through the [thread](https://github.com/TriForceX/MiyooCFW/issues/314) that was opened recently where we discussed this.

RetroArch has its own compatibility list here: https://docs.libretro.com/library/gpsp/#compatibility

Rezzy-dev, 2022-04-22:
I'm working on optimising @gameblabla's port of VBA Next (https://github.com/gameblabla/vba-next-portable) to be usable on Miyoo.

FloofTheWolf, 2022-04-23:
So,

Tried the shutdown shortcut when the crash/freeze happened.

It unfortunately didn't work.

I can say that I was playing "Pokemon - Emerald" if that can be any help.

I understand that fixing bugs is very hard so I hope I don't really rush the people working on this.

Rezzy-dev, 2022-04-23:
> It unfortunately didn't work.

I've never heard of the quick shutdown shortcut/hotkey not working before. :confused: That would indicate a whole system freeze. Are you sure you're pressing the right button combination?

You need to hold down the Reset button (which is at the top of the Pocket Go and the minus-shaped button on the PowKiddy Q90), and press the Select button once. (You don't hold down the Select button, just press it once while holding down the Reset button.)

jSTE0, 2022-04-23:
Try the same title with the gpSP core in RetroArch. If it crashes/hangs there too, then try it with RetroArch on a different platform (Windows/Mac/Linux). If it also crashes/hangs, that suggests an issue with the emulator core itself and you should [report it upstream](https://github.com/libretro/gpsp/issues). If not, then it could be platform-specific and someone here might be able to assist.

Rezzy-dev, 2022-04-23:
If the shortcut still doesn't work, that would indicate a hardware-side fault with you device/unit. Then the freeze is most likely not caused due to a bug in the emulator, but rather a hardware problem with the components in the device, such as the RAM (memory), or the SD card. Only such a fault is likely to cause a system-wide freeze on Linux/Miyoo.

FloofTheWolf, 2022-04-25:
Just remembered,

I have tried reinstalling the software on the SD card(an 16 gig SanDisk one) A couple (like 3-4) times. This issue was happening before so I thought that reinstalling it might fix the problem. Cloud this make some kind of an corruption or other problem instead? The card is from a pretty good brand so I doubt it would corrupt by itself (if that is the problem).

Also I'm using the powkiddy v90, if that makes any difference.

jSTE0, 2022-04-25:
Unlikely, if it's consistently crashing with the same title and emulator. Emulators not compatible with all possible titles, especially one that's not cycle-accurate like gpSP. Try it on a different emulator first.

Rezzy-dev, 2022-04-25:
Re-flashing the SD card (re-installing the system) will not cause such a problem. What I'm referring to is a physical fault with the SD card itself, potentially. You could try using a different SD card. But if we're talking about hardware failure, then the memory (RAM) is the more likely culprit here.

If the RAM is faulty, random system freezes do tend to occur while the RAM is being used, while the device is running.

You weren't able to shut down the device with the hotkey, even after following the instructions carefully?

FloofTheWolf, 2022-04-26:
I was doing it wrong. I was holding all of them instead of releasing the start. So I shut it down in the menu. It worked. BUT when I tried it on "Mario & Luigi superstar saga" ( a game that constantly crashed in the middle of the "star fly into screen" part) it did work (the screen became black and the backlight turned off) when I flicked the switch back down then up, to turn the device on it showed me the "unclean shutdown detected checking fat32 partition" screen. 

So that means that it didn't make a clean shutdown, even though it works on the menu screen. Also the aftermentioned title crashes much more "predictably" than all of the other ones which have crashed.

Rezzy-dev, 2022-04-26:
> I was doing it wrong. I was holding all of them instead of releasing the start. So I shut it down in the menu. It worked. BUT when I tried it on "Mario & Luigi superstar saga" ( a game that [...]

It sounds to me like you're still not following the steps correctly. Start has nothing to do with the quick shutdown. You need to **hold Reset** and **press Select**.

Holding Start is a quick way to bring up the regular shutdown menu in the main menu (GMenuNX) of the system.

Pressing (or holding) just the Reset button in the main menu (GMenuNX) will put your device into sleep/suspend mode -- which is **not** a shutdown unless you leave the device sleeping for a longer time (10 minutes). In this instance you will see a sleep/suspend message before the screen turns off.

It's also possible that your SD card got corrupted from the many unclean shutdowns/poweroffs. If you're getting an "unclean shutdown detected" message every time you turn on the device, no matter how you shut it down, then that indicates a corrupted file system on the SD card. Then you will need to re-flash the card (re-install the system).

FloofTheWolf, 2022-05-04:
Sorry! I messed up the start & select buttons. I was actually pressing reset + select.

Rezzy-dev, 2022-05-04:
I've never hears so far of Reset + Select not working. Holding down Reset and pressing Select sends a system-wide command to force close any running programs and shut down the system.

If it doesn't work, that means that not only the emulator is frozen, but the whole system! That indicates a bigger issue, usually with the hardware -- as I said (particularly the RAM/memory or SD card). Especially if it happens while running other programs, too.

Rezzy-dev, 2022-05-05:
Occasionallly, certain dysfunctional scripted system commands can cause such a system freeze, but I take it that you haven't been tinkering with the system scripts/files, right?

FloofTheWolf, 2022-05-05:
I haven't done anything with the files except for adding some games, so it probably is the problem with the aftermentioned ram/SD card. Now if the card is somehow faulty, that isn't such a big problem, I can just get a new one, but the if the ram is problematic, that kinda worries me, beacuse it's probably not that easy to fix, if it is possible to fix in the first place

Rezzy-dev, 2022-05-05:
Yep. You should try installing the system on a new SD card. That way you'll know if it's the SD card at fault (and you can always use a spare card anyway). Freezing due to the SD card would take place while the system is writing to he card (saving), usually. If there's anything faulty about the card, it will most likely show up during the writing process (or occasionally when loading data, if the data held on the card is corrupted).

Another thing you can do is hit the team up on the Discord channel, and ask someone to test that exact ROM with the emulator for you on their device -- to see if it freezes for them, too, and if they can shut it down with Reset + Select when that happens. This will show if the problem is software-side.

As a last resort, if it's your device's RAM that faulty, it's not impossible to replace the RAM, but it'll be much easier if you can return the device to the seller for being faulty and get a refund (not sure when you bought it). Just make sure that it really is the RAM/device that's faulty -- not the SD card, and not something software-side.

FloofTheWolf, 2022-05-10:
At the moment I don't have access to my pc, so in turn I can't flash the card. Also I have the device for like a year now, so returning it is thrown out of the window.

When I will be able to flash the card I will update you on the issue.

But for now, I will try to reach out to some people on discord, so they can test the software out on their devices.

Also, cloud the roms themselves be faulty, to the point that they cause the crashes?

Rezzy-dev, 2022-05-11:
> Also, cloud the roms themselves be faulty, to the point that they cause the crashes?

If the freeze happens only with certain ROMs, then yes. A combination of a corrupted or badly dumped ROM in addition with bad memory management/handling in the software/emulator can overfill the RAM memory can cause such a system freeze. *Nods.* It's rare, but it's possible.

eknives, 2023-01-12:
> I'm working on optimising @gameblabla's port of VBA Next (https://github.com/gameblabla/vba-next-portable) to be usable on Miyoo. If successful, that will improve GBA game compatibility greatly on the platform (and performance over the RetroArch mGBA as well).

It would be great! Please do it.

Apaczer, 2023-03-15:
Compatibility thread for gpSP: https://github.com/TriForceX/MiyooCFW/discussions/482
Updated hotkey's list with software/system shutdown feature: https://github.com/TriForceX/MiyooCFW/wiki/Miyoo-CFW#console-hotkeys

closed 2023-03-15 by Apaczer