TriForceX/MiyooCFW #371: [pcsx rearmed] [pocketgo1] The emulator wont render on fullscreen

tatinnn, 2022-04-29:
When using the standalone version of pcsx rearmed wich comes included with the cfw 1.3.3, all games will render with black bars at the top and bottom of  the screen. When using the retroarch version of pcsx rearmed it will use all of the screen. Since the device im using (pocketgo) has a 4:3 screen, it means the retroarch version is rendering the games correctly but the standalone is not. Is there some way to tinker the emulator so i can utilize this small screen to the fullest?

Rezzy-dev, 2022-04-30:
Are you using PAL or NTSC ROMs? You should be using NTSC. Miyoo devices don't support screen scaling (for lack of a graphics card, the CPU would need to do all the scaling work, which is not useful/practical).

Apaczer, 2023-03-15:
The PS1 games may have different resolution from game to game & even within one game! 
Full screen aspect 4:3 scaling on RetroArch will "forcfully" scale-up those smaller or uneven 4:3 ones to fit your screen.

For reference: https://docs.google.com/spreadsheets/d/1UgysgrgqbiIlyHIiwCxVoWMu1bwgO2OBlDO1ORpsi78/edit#gid=0