TriForceX/MiyooCFW #372: PCSX REARMED

ShyCode-analyst, 2022-05-02:
I want to know if I'm on the best version for performance. The emulator runs fine I put 1 frameskip, and it works very well.

I have all the plugins installed in the emulator folder, I saw that in the emulator options to select the plugin manually, some of them can change the performance?

About the CPU clock, I don't understand it too, it comes with a default of 50%, changing that affects what?

currently playing testing Valkyrie profile
After a while playing the emulator crashes less, but it all starts crashing and stabilizing... Is there any way to change this?
=====
Build: Oct 28 2019 r22 88 g3c89130
PCSX - Reamed.
=====
SD Card 8GB 
SDHC
4GB/s
SanDisk
=====
miyooCFW

DoctorOey, 2022-05-04:
Powkiddy v90 not design for 3D psx games.
Cause it not have 3d accellerator /graphic card.
Do not force your device it may make your chipset burn.
U can use configuration (cfg) in this thread of miyoo project.

ShyCode-analyst, 2022-05-04:
Valkyrie profile is in 2d

I solved puting GPU  = Turn Off light, Works well