TriForceX/MiyooCFW #396: Battery level indication

Rezzy-dev, 2022-06-19:
Are we certain about these numbers?

[ gmenu2x.cpp ]:
```
uint16_t GMenu2X::getBatteryLevel() {
	int32_t val = getBatteryStatus();

if (confStr["batteryType"] == "BL-5B") {
	if ((val > 10000) || (val < 0)) return 6;
	else if (val > 4000) return 5; // 100%
	else if (val > 3900) return 4; // 80%
	else if (val > 3800) return 3; // 60%
	else if (val > 3730) return 2; // 40%
	else if (val > 3600) return 1; // 20%
	return 0; // 0% :(
}
```

Because my PowKiddy Q90 doesn't seem to ever show the AC icon while charging (instead it shows 100%, 5), and once charged and disconnected from the charger, it shows the 4th icon, 80% on full charge, not 5.

I'm wondering if it's my battery (it's a new device), or if the calculation/estimation in the code is wrong here.

Rezzy-dev, 2022-06-19:
```
// return 5 - 5*(100-val)/(100);
	return 5 - 5 * (max - val) / (max - min);
```

What on earth is this doing, at the end of the function? Why "5 -"?  Shouldn't it be "6 -"? :eyes: And what happened "val" being reported as less than zero on charging? This may be the source of our bug.

jSTE0, 2022-06-19:
> This may be the source of our bug. thinking

No, the first bit that matches `"BL-5B"` already returned early. The battery type is simply [hardcoded](https://github.com/MiyooCFW/gmenunx/blob/b5642412cd6d5478f71fc38a7cafa050b509de81/src/gmenu2x.cpp#L1365). FWIW the V90 actually comes with a BL-5C (it's a Nokia battery model designation).

I've never seen an AC icon and mine reads 4230 fully charged (so it displays the icon with all bars), with or without the charger plugged in. RetroArch uses a different table derived from the author's V90 libretro/RetroArch@9dd034e5eaeb.

Rezzy-dev, 2022-06-20:
Excellent! That means that for whatever reason this line is never triggered:
```
if ((val > 10000) || (val < 0)) return 6;
```
A return of 6 gives us the AC icon.

While plugged into charging, what does your device read? This is where we're running into something unexpected. The code expects the device to read over 10000 or a minus value while it's charging. I'll check mine (my PowKiddy Q90), too.

Rezzy-dev, 2022-06-20:
Hmm... My PowKiddy Q90 does not seem to be giving off any such indication that it is plugged in and charging. It simply reads the current charge of the battery. This is why we never saw the AC icon.

I'll wait until it charges fully, and check the value then.

Rezzy-dev, 2022-06-20:
4230 fully charged -- the same, you're right. And I was in error, the 100% icon does show on a full charge, it's just that my PowKiddy's battery drops to below 4000 rather quickly.

So the indication is working correctly, except that the AC icon (6) condition is never true. It seems the current implementation of hardware and system software doesn't register and record when the device is plugged into power, so this feature is missing and, consequently, the AC icon never shows.

Rezzy-dev, 2022-06-20:
Not sure to leave this open as a bug (missing feature) or close it. :thinking: I'll leave it in case we want to reimplement the AC icon, and if there's a way to with the current hardware.

jSTE0, 2022-06-20:
The kernel is reading the KEYADC Data Register, multiplying that by 90, and reporting that as `voltage_now`. That register is a 6-bit value so the largest value that can be reported is `63 x 90 = 5670`, which is less than 10000.
It's a little odd that the battery voltage would do double-duty as a charging indicator as devices typically report them separately.

Rezzy-dev, 2022-06-21:
Yeah. It's strange programming with that return 6. Especially given that it's a 6-bit value that's being stored in that file. But I guess it all depends on the programming of the software that reads and reports that hardware status.

I agree that it's quite odd for it to do a double-duty, but I've seen this done in other system software before, so it wouldn't be the first time. Technically, while the device is plugged into power, the voltage reading has little significance -- so it's not unreasonable to drop it for the duration and signal with a negative value, for instance, that the device is plugged into power instead.

I poked around in the "sys" files there to see if there was any record of the device being plugged into power in any of the other files -- something we could use for triggering the AC icon -- but no, I couldn't find anything. :/

MiyanoOsu, 2022-07-23:
@Rezzy-dev BL-5B is name of battery and it just a name, do nothing, i mod battery to li-ion 5000mAh and charger work fine without mod kernel, i use it for 6 months. 
Folow this code, it detect current voltage of battery. It will better if have a driver to checking voltage chager. But you can try this code as a alternative menthod. 
I write it depend on how voltage charging, first battery full charge at 4.2v it mean 4200, current voltage when charging alway higher than before.
[image]

Change this line: 
if ((val > 10000) || (val < 0)) return 6;
To:
 if ( (val < 4200) && (val > val+10)) return 6;
It may work!!!

Wilter, 2022-08-07:
> ```
> // return 5 - 5*(100-val)/(100);
> 	return 5 - 5 * (max - val) / (max - min);
> ```
> 
> What on earth is this actually doing, at the end of the function? Why "5 -"? Shouldn't it be "6 -"? 👀 And what happened to "val" being reported as less than zero on charging? This may be the source of our bug. 🤔

From 0 to 5, there are 6 values.
The function's return is an integer.
There are 6 defaults statuses, you can see the battery icons to see that (example "...\gmenu2x\skins\OldBoy\imgs\battery").
I think the seventh status, AC status, is because the battery has been removed and is only charging.
This return is practically a mapping, for integer values from 0 to 5, it is mapping the percentage (0 to 100) in these values there.

Rezzy-dev, 2022-08-07:
Ah, it's assigning the percentage based on the charge value provided. It only needs to calculate 6 values, as the 7th is the charge icon, for which the charge value is symbolical/nonsensical. It needs not calculate and display the 7th value in percentage. Now I understand. :+1: 

I've been thinking about this, and it would make much better sense for the charge icon to be independent of the actual voltage inside the unit. There should be a driver registry/indicator whether the AC is plugged in or not. The charge icon display should be hooked up to this record/indicator, and independent of the voltage reading (percentage displayed). But I suppose this is a driver/hardware question, in whether this is technically feasible -- one of which I don't know much about. :thinking:

Rezzy-dev, 2023-02-05:
I just realised there is a way to implement this!

We just need to run a periodic loop/function to check to see if the battery "voltage" is increasing: store the current voltage value in a variable, and compare that a few ticks later with the voltage value now. If the voltage value has _increased_, show the charging icon (return 6). If the voltage value drops check to see which icon to display per current state of the battery (return 1 to 5).

This means that in practice, after the device is plugged in, a short interval later the charging icon turns on (when the device is actually charging). And a short period after the device is disconnected, the charging icon will turn off and change to show the current charge level of the battery.

I'll have a look into implementing this a bit later -- if someone doesn't beat me to it (you're welcome to). :)

szymor, 2023-02-23:
@tiopex slight suggestion - maybe "charging" instead of "charge"? "Charge" suggests charge stored in the battery measured in Wh.

jSTE0, 2023-02-24:
On a laptop, the file `/sys/class/power_supply/<battery-name>/status` has one of these values:
```C
static const char * const POWER_SUPPLY_STATUS_TEXT[] = {
        [POWER_SUPPLY_STATUS_UNKNOWN]           = "Unknown",
        [POWER_SUPPLY_STATUS_CHARGING]          = "Charging",
        [POWER_SUPPLY_STATUS_DISCHARGING]       = "Discharging",
        [POWER_SUPPLY_STATUS_NOT_CHARGING]      = "Not charging",
        [POWER_SUPPLY_STATUS_FULL]              = "Full",
};
```
from [the 6.2 source](https://github.com/torvalds/linux/blob/c9c3395d5e3dcc6daee66c6908354d47bf98cb0c/drivers/power/supply/power_supply_sysfs.c#L75-L81.)

tiopex, 2023-02-28:
After investigation, not all devices supports charging information. Currently known device is sup m3.
Pocketgo and v90 are not supported. 

The driver will return in file /sys/class/power_supply/miyoo-battery/status below values:
- Charging
- Discharging
- Unknown

Rezzy-dev, 2023-06-16:
Awesome work @tiopex! ❤️

It's unfortunate that not all firmware support this feature. In that case, perhaps the fallback solution I suggested (https://github.com/TriForceX/MiyooCFW/issues/396#issuecomment-1417871771) can be used, since it covers all devices? It may not give us second-accurate charging state reporting, but it still gets the job done, and on all devices.

For the devices that do not support this feature, does the reported status stay "unknown"?

If so, then we could use your driver reporting implementation first and foremost, and if the reported status stays "unknown", then fall back to my suggested workaround to see if the battery charge has increased over time. (If it has, then show the charge icon.)

That would give us the best of both worlds -- both speed to the reporting, and cross-device compatibility/reliability.

MiyanoOsu, 2023-06-17:
Sadly, it is not possible for fall back solution, when charging, charger apply a spectific voltage to battery, when it reach this voltage then new spectific voltage will apply it will do like this until it reach the lastest voltage and stop charging, in a cycle of charging battery chage voltage about 5 times or less, as i see status report by my q90. 
Also it take a long time to report new battery status, so in midle of time for new status report, what to prove that battery is charing? When unplug chager there is no info report that it unplug.
It will easy when otg support, just have a code detect line coding from otg pin usb dp and usb dm, but adapter charger not support it, maybe solution is modify hardware by wire gpio pin that not use to usb vcc 3.3V via resistor. Then firmware will detect 0/1 on this gpio pin.

tiopex, 2023-06-17:
> For the devices that do not support this feature, does the reported status stay "unknown"?

Yes

Rezzy-dev, 2023-06-17:
> Also it take a long time to report new battery status, so in midle of time for new status report, what to prove that battery is charing?

The charging icon only needs to show for as long as the device is still charging (not whether it's plugged in). So if the voltage is not increasing anymore (if it stays the same or decreases at the next interval) the charging icon can be dropped.

Ideally, we want the charging icon to show as soon as the device is plugged in and to be dropped when the device is charged or disconnected from the charger, but as a fallback solution, I think the method I suggested would work in practice.

The question is how large this interval actually is -- how often the battery charge level report gets an update by the driver. If it's something like a few seconds, then my fallback method will do just fine. If it's something like a few minutes, then it won't work so well.