TriForceX/MiyooCFW #413: protect save data please

tetradog, 2022-08-21:
i need a solution for sd data corruption because i lose many save datas...
i follow all steps to system power off but it's hapens again...(the card is a new one)

my suggestion:
1-I think if it's possible make a partition with the lastest save data to protect others partitions with data...
once the player bring back to menu and init shutdown program this program has to sync the save data :D 
2-A backups partition with manual start with battery level check is very important to me because save data has many hours spended.

Rezzy-dev, 2022-08-22:
The backup on the SD card will not help if the SD card is faulty.

I suggest that you periodically back up the contents of "main" on your SD card to your computer. Then you can easily restore all your data in the chance that the SD card gets corrupted and you need to re-flash it.

tetradog, 2022-08-28:
data corruption affect others partitions?

Rezzy-dev, 2022-08-29:
It depends on what the cause of the corruption is. It could be an uncrean shutdown while the system was writing to the card, software-side (this is the most common smaller issue, usually easily and automatically fixed), or it could be a physical fault with the card (this is the more common larger issue, that cannot be fixed software-side).

Most SD cards have a limited life expectancy of a certain amount of write cycles (usually around 100,000). They do not last forever. When it reaches its end of life, data on the card is simply lost, and the card becomes practically unreadable. And many SD cards ship faulty.

If it's a physical fault with the card, backing up to another partition on the card will not save your data from being corrupted.

nfriedly, 2022-09-01:
> data corruption affect others partitions?

On flash storage, yes. Internally, data is spread across multiple flash cells to speed up reading and writing, but this means that if corruption occurs anywhere, it can affect data all over the drive.

tetradog, 2022-09-01:
> > data corruption affect others partitions?
> 
> On flash storage, yes. Internally, data is spread across multiple flash cells to speed up reading and writing, but this means that if corruption occurs anywhere, it can affect data all over the drive.

thanks, its very instructive :)