TriForceX/MiyooCFW #417: Toolchain Fails to Build

Rezzy-dev, 2022-08-25:
The buildroot fails to build, now (using Debian 9, same as before):
[image]

TriForceX, 2022-08-27:
Which one are you using?

Rezzy-dev, 2022-08-27:
https://github.com/MiyooCFW/buildroot.git

aksenav2, 2022-08-29:
[image]

Linux mint and ubuntu

nfriedly, 2022-09-01:
I was chatting with @aksenav2 on slack, but figured I should mention this here also:

@Rezzy-dev https://github.com/MiyooCFW/buildroot.git is configured to use an external toolchain so that it doesn't have to keep recompiling that, and can just build the rootfs. Do you have the toolchain in `/opt/miyoo`? See https://github.com/MiyooCFW/toolchain/releases/tag/v2.0.0

(And, normally I don't put the buildroot source in `/opt/miyoo`, although I don't think that's the source of the problem.)

nfriedly, 2022-09-01:
I just added a couple of notes to https://github.com/MiyooCFW/buildroot#miyoocfw-buildroot clarifying that 

1) We're not actually using that repo's outputs currently, and
2) It expects the toolchain to be downloaded ahead of time

Rezzy-dev, 2022-09-01:
Why the need for all these additional environment variables to be configured manually in the terminal before compilation?

```
CROSS_ROOT=/opt/miyoo
CROSS_TRIPLE=arm-buildroot-linux-musleabi
SYSROOT="${CROSS_ROOT}/${CROSS_TRIPLE}/sysroot"
PATH="${CROSS_ROOT}/bin:${SYSROOT}/usr/bin:${PATH}"
ARCH=arm
CROSS_COMPILE="${CROSS_TRIPLE}-"
```

Why aren't they noted/pre-defined in the Makefile?

Also, this is an entirely new build process that has not been documented in our main guide, the Wiki:
https://github.com/TriForceX/MiyooCFW/wiki/Build-Source
https://github.com/TriForceX/MiyooCFW/wiki/Making-Games

What happened to the previous toolchain that used to occupy this URL/space? Things are getting very messy, and hard to follow... :frowning_face:

Rezzy-dev, 2022-09-02:
> Do you have the toolchain in `/opt/miyoo`?

And yes, I've had the toolchain at `/opt/miyoo` (but we've also tried compiling it elsewhere, without success).

nfriedly, 2022-09-02:
I went through the various repos we have and that set of variables seemed to be what they needed to be able to compile things for Miyoo.

The wiki hasn't been updated partially because this is still a work in progress - we still can't build an image fully from source. And partially because I don't have edit permissions for the wiki.

I don't think that that our buildroot repo was ever used for 1.3.3, the tool chain for that was a zip file in Google drive.

nfriedly, 2022-09-02:
What does `stat /opt/miyoo/bin/arm-buildroot-linux-musleabi-gcc` output?

What about `ls /opt/miyoo/bin/`?

Rezzy-dev, 2022-09-02:
> I don't think that that our buildroot repo was ever used for 1.3.3, the tool chain for that was a zip file in Google drive.

No, that is not correct. This repo/URL was previously used for building software for 1.3.3, musleabi, as you can see in the Wiki. 

The [pre-packaged archive](https://drive.google.com/file/d/1Zp6elbB3r4Hq7G7FKXNtBZMTz0H93RiY/view?usp=sharing) is for the uclibcgnueabi build.

Rezzy-dev, 2022-09-02:
> @nfriedly: I don't have edit permissions for the wiki.

@TriForceX

nfriedly, 2022-09-02:
Did anyone successfully use our build route repo to build software for 1.3.3?

Regardless, the reason I switched to an external tool chain was that it *massively* reduced build times.  (I'm trying to get things to the point where CI can build complete images from source, so speeding up the build times really helped there.)

nfriedly, 2022-09-02:
Oh, sorry if I missed things up for you. 

I think the only thing that needs to change in the wiki is that it should point to the toolchain repo instead of the builroot repo. They're both the same version of buildroot, just with slightly different configurations. The toolchain one is set up for building the SDK, and then the buildroot repo is supposed to use that to build a root filesystem.

Rezzy-dev, 2022-09-02:
That's okay. Which is the toolchain repo? Where can I still find the 1.3.3 musleabi toolchain/SDK?

I can change the URL in the Wiki (update the documentation).

nfriedly, 2022-09-02:
This is the toolchain repo: https://github.com/MiyooCFW/toolchain

Rezzy-dev, 2022-09-02:
Just to confirm, the toolchain needs to be in `/opt/miyoo`, now, to successfully build?

nfriedly, 2022-09-02:
Yeah, I don't think that changed (?)

Rezzy-dev, 2022-09-02:
Thanks @nfriedly! :hugs: 

Successfully built, and the 'Making Games' documentation has been updated accordingly.

I take it that the 2.0 toolchain/SDK is backwards compatible with building for the 1.3.3 system?

closed 2022-09-02 by Rezzy-dev