TriForceX/MiyooCFW #431: Miyoo Mini V2 screen refresh problems

vanfanel, 2022-09-27:
Hi there,

I already asked on the Onion OS issue tracker, but it seems to be a problem with the console fw.

The problem comes down to this: RetroArch VSYNC does not work. One can easily notice scrolling problems on every emulated system using scrolling test included in the 240p test suite: https://artemiourbina.itch.io/240p-test-suite

Scrolling presents periodic hiccups, which are really fast tearing problems caused by the lack of working vsync. Even RetroArch scrolling text movement is wrong. 

I suspect the dingux_sdl backend in RetroArch, but who knows. I know everybody should be able to see the problem, but I find rather strange that nobody mentions it.

szymor, 2022-09-27:
MiyooCFW is not supported on Miyoo Mini V2. You should not create issues here for that platform.

Said that, I am afraid that firmware for Miyoo Mini V2 is not released as open source, so it would be hard to fix that for third party developers.

closed 2022-09-27 by vanfanel

szymor, 2022-09-27:
Firmware is not developed publicly, so I guess you need to inform the manufacturer about the issue.
@TriForceX correct me if I am wrong :)