TriForceX/MiyooCFW #432: firmware wont use all my sd card space

wind013, 2022-10-03:
https://i.imgur.com/8MUMjMu.png

And windows wont let me resize the main partition for some reason and if I try to create a new partition windows gives me error "The parameter is incorrect"

How to fix this? I want to play my games please help me :(

wind013, 2022-10-03:
managed to get it fixed by resizing main partition with the partition wizard tool

closed 2022-10-03 by wind013

Rezzy-dev, 2022-10-03:
(This is a support request, not a bug report. Moving it to discussions.)

You'll need to use a tool for Windows like Partition Wizard to resize the partition labelled "main" on the SD card. Download links and instructions here:

https://github.com/TriForceX/MiyooCFW/wiki/Custom-Firmware-Setup