TriForceX/MiyooCFW #436: i cant dounload

crobatman0, 2022-10-15:
evry time i download the software i git to 80% of the way then im told i need the websites premishan to download it i dont know how to fix this

Rezzy-dev, 2022-10-16:
:open_mouth: Which download link?

crobatman0, 2022-10-16:
> 😮 Which download link?

the powkiddy v90

Rezzy-dev, 2022-10-16:
Works for me. Try emptying your web browser's cache, quitting the browser, and downloading it again. (Or use a different computer/web browser to download it.) Maybe something went wrong in your web browser when downloading it for the first time.

If the permission thing shows up again, take a screenshot and post it here. It may help tell us what is wrong.

crobatman0, 2022-10-16:
> Works for me. Try emptying your web browser's cache, quitting the browser, and downloading it again. (Or use a different computer/web browser to download it.) Maybe something went wrong in your web [...]

[image]

crobatman0, 2022-10-16:
> Works for me. Try emptying your web browser's cache, quitting the browser, and downloading it again. (Or use a different computer/web browser to download it.) Maybe something went wrong in your web [...]

thats what it is telling me i cant find anything else i can do i have tryed on 2 difrant computers both brave web browser and chrome clearing evrything i have tryed downloaded it about 5 times

Rezzy-dev, 2022-10-17:
:open_mouth: That's so strange! Could it be that the download has completed, and your computer is actually asking for authorisation to open the archive? (That would make a lot more sense.)

Can you actually extract/open the downloaded archive?

Rezzy-dev, 2022-10-17:
https://www.technipages.com/chrome-download-failed-insufficient-permissions

Looking through Google, it seems to be a common issue/bug with the Chrome web browser.

Try the following things:

1. Switch to and download in Incognito mode (this will disable all the addons that may be interfering for the duration of your download session).
2. Empty your browser's cache and quit the browser before you attempt the download again.
3. Make sure you have enough free space on your hard drive to download the archive (at least 4 GB), and try changing the save location.

Apaczer, 2022-10-17:
Try github release link: https://github.com/TriForceX/MiyooCFW/releases/download/1.3.3/powkiddy-q90-v90-cfw-v1.3.3.7z

Rezzy-dev, 2022-10-17:
I don't think it's related to the download source. Looking at the Google search results, it's most likely a bug in the browser (or the browser working with Windows). Since it happens after the 80% mark, I'm thinking the browser is failing to move the download from the cache to the chosen destination on Windows.

Switching to a different browser would likely do the trick of escaping this bug. But yeah, one of these links may work, too. But in either case, I'm pretty sure it isn't a problem with the Github server. That "permission" error message is Google's way of hiding their own mistake (they do that a lot). Google's the next Microsoft. 🙄

Rezzy-dev, 2022-10-17:
On the server-side, the only thing that could happen is a timeout after a long period on the download. But somehow I really doubt this is the case here. You have a relatively fast Internet connection, right, @crobatman0? How long does it take for you to download the archive? Is it more than 30 minutes?

nfriedly, 2022-10-17:
Try this link: https://nc.home.nfriedly.com/s/zc3N9T7Q7828XiH/download?path=%2F&files=PowKiddy_Q90_V90_Only_CFW_v1.3.3_04-10-20.img.7z

nfriedly, 2022-10-17:
You could also try this utility, I don't really know what it does but it did fix one computer I have that had trouble with a lot of random downloads: https://www.speedguide.net/downloads.php

closed 2022-10-19 by Rezzy-dev