TriForceX/MiyooCFW #437: Error build tolchain

aksenav2, 2022-10-21:
Hello, after reinstalling the system, I decided to build again  tolchain , but again it does not work, an error is displayed, what could it be? 
[image]

Rezzy-dev, 2022-10-21:
Did you follow these instructions?

https://github.com/TriForceX/MiyooCFW/wiki/Making-Games

Mae sure you're compiling the toolchain source, not the buildroot one (the buildroot is currently under construction).

Rezzy-dev, 2022-10-21:
Also note that you have pre-built downloadable packages of the toolchain now:

https://github.com/MiyooCFW/toolchain/releases

 You don't _need_ to build from source.

aksenav2, 2022-10-21:
> Did you follow these instructions?
> 
> https://github.com/TriForceX/MiyooCFW/wiki/Making-Games
> 
> Mae sure you're compiling the toolchain source, not the buildroot one (the buildroot is currently under construction).

Yes, I did exactly according to this instruction. Worked out well in the past

aksenav2, 2022-10-21:
> Did you follow these instructions?
> 
> https://github.com/TriForceX/MiyooCFW/wiki/Making-Games
> 
> Make sure you're compiling the toolchain source, not the buildroot one (the buildroot is currently under construction).

I followed these instructions.  sudo mkdir /opt/miyoo
cd /opt/miyoo
sudo git clone https://github.com/MiyooCFW/toolchain.git
cd toolchain
sudo make sdk

Rezzy-dev, 2022-10-21:
Yep, that should do the trick. Not sure what you're doing with the PATH export there. You could try it without that.

Rezzy-dev, 2022-10-21:
The error it's throwing is that it's missing a C(++) main header file. Make sure you have the full GCC compiler installed on your system. Since this is a fresh system installation, have you installed all of the required C(++) development environment on your system?

szymor, 2022-10-21:
Refer to the buildroot manual to check if you have all needed dependencies installed.

aksenav2, 2022-10-21:
> The error it's throwing is that it's missing a C(++) main header file. Make sure you have the full GCC compiler installed on your system. Since this is a fresh system installation, have you installed all of the required C(++) development libraries/environment on your system?

Indeed, due to my inattention, I forgot to set package g++ that's the whole problem. Thank you very much for your help and sorry for the inconvenience.

aksenav2, 2022-10-21:
> Indeed, due to my inattention, I forgot to set package g++ that's the whole problem. Thank you very much for your help and sorry for the inconvenience.

New error
[image]

Rezzy-dev, 2022-10-21:
That looks like an actual error in the programming/code.

Any particular reason you're compiling this from source? Just use the pre-built toolchain -- it saves headaches.

szymor, 2022-10-21:
The last error is caused by libc (as far as I remember) update which causes incompabilities. I think you can easily find a proper patch in the internet which needs to be copied to a proper package folder (maybe with slight modification) - I will not upload it here because I do not have access to my home laptop at the moment.

szymor, 2022-10-21:
Ok, I checked my buildroot repo. It seems I have updated m4 to 1.4.19. Do not forget to update hash in the package folder.

szymor, 2022-10-23:
@aksenav2 I fixed the issue and managed to build the toolchain properly. Have a look at https://github.com/szymor/buildroot-miyoo/commit/4c5e6e813556d3c6fb37f8c290b5447ef5bca963

closed 2022-10-28 by Rezzy-dev