TriForceX/MiyooCFW #441: How to re add a section

CombatHit, 2022-11-01:
I am on a powkiddy q90 and accidentally deleted the emulator section. All the files are still on the sd card the section just doesn't show. How can I add them back?

Rezzy-dev, 2022-11-01:
The physical files are still there, but the launchers have been deleted. My advice is to be careful when deleting things. Think twice before you do -- and many times you don't need to: simply moving the files to a different location on the SD card (or onto your computer) is enough (even when you need to free up space). Opt to archive rather than delete.

That said, here are the files you've deleted (in their original publication/release):

[emulators.zip](https://github.com/TriForceX/MiyooCFW/files/9914588/emulators.zip)

Unpack this and copy the emulators folder with its launchers to: [main:] gmenu2x/sections/

CombatHit, 2022-11-01:
Thanks a lot