TriForceX/MiyooCFW #449: Powering off the correct way results in unclean shutdown

nap-arts, 2022-12-09:
I power off the device via the main OS menu every time, and every time it boots back up it says it was an unclean shutdown. One thing of note is that when I confirm the shut down from the menu, the yellow LED on the device stays on indefinitely, so I suspect the device is getting stuck somewhere in the powering off process.

Rezzy-dev, 2022-12-09:
Which device?

It's normal for the light to stay on, because shutting down the system does not power off the device. You still need to flip the switch to turn off the device at the end of the shutdown process (once the screen has turned off).

If the unclean shutdown persists on each boot, even after a clean shutdown, that suggests that something in the file system got corrupted, and can't be fixed. You may need to re-flash the SD card.

Also be sure that the SD card you are using is NOT the one that came with the device. The SD cards that are shipped with the devices are generally faulty ones. Use a new/different SD card.

nap-arts, 2022-12-11:
> Which device?
> 
> It's normal for the light to stay on, because shutting down the system does not power off the device. You still need to flip the switch to turn the device off at the end of the shutdown process (once the screen has turned off).
> 
> If the unclean shutdown message persists on each boot, even after a clean shutdown, that suggests that something in the file system got corrupted and can't be fixed. You may need to re-flash the SD card.
> 
> Also be sure that the SD card you are using is NOT the one that came with the device. The SD cards that are shipped with the devices are generally faulty ones. Use a new/different SD card.

Thanks a bunch!