TriForceX/MiyooCFW #451: Not rushing but thought I'd ask.

arrow1700, 2022-12-23:
Was hoping to ask if you guys are close to have 1.4.0 update for the v90 out. I'm patient and can wait but was curious. Im not rushing you guys as I know that Rushing people can cause issues. I'd rather have something optimized in something that was rushed to put together. I was just curious if you were close. And I apologize if this is not the area to ask not too familiar with GitHub.

szymor, 2022-12-23:
@TriForceX needs to answer this question as I am not sure what kind of features are planned for the next release. I would say screen tearing is fixed though it causes performance drop by ~20%, so it is not ideal. Probably one setting entry needs to be added to make possible to switch it on and off. Anyway, huge thanks to @tiopex for working on that feature, he pushed CFW into high quality gaming grounds.
The other big feature (at least from dev's perspective) was to clean the build environment and separate emulators and ports from the firmware itself. I think this goal has been achieved, too. It will allow to update CFW libraries more easily from now on, so the devs can use the latest versions of dependencies.
Have in mind that we all work in our free time, so it is hard to say about any specific deadline. To be honest, the staff changes from time to time, so it is hard to plan ahead in such an environment. For example me - I mostly retired from porting anything to MiyooCFW. The same goes for some other team members.

arrow1700, 2022-12-23:
> @TriForceX needs to answer this question as I am not sure what kind of features are planned for the next release. I would say screen tearing is fixed though it causes performance drop by ~20%, so it [...]

....
Thank you for replying I didn't want to come off as though I'm rushing the devs so I figured I'd try to implement in that in it that I wasn't intending that I was just curious if it was close
Like I said I'd rather have something that's optimized than something that was just slapped together so I'm willing to be patient still. So I'm not rushing ya. Anyways thank you for replying.

TriForceX, 2022-12-23:
@szymor is right, some stuff is still very experimental, we dont want release something unstable, so we are trying to make the experimental features toggleable so people doesnt have to complain about the performance issues.

The big work here was to clean the mess with the project and source code, and as been said, the few devs who help with this project does the stuff by time to time in their free time, sometimes few code lines per month (lol), theres no estimated time or such things due the lack of support, thats the true, always just 3 or 4 devs were helping on this project and they have their own real life stuff, theres no a team behind this, just few people doing things on their limited free-time.

PS: The cfw status always have been here: https://github.com/TriForceX/MiyooCFW/discussions/231

closed 2022-12-23 by TriForceX