TriForceX/MiyooCFW #455: Q90 bad mapping for G&W emulator

eknives, 2023-01-09:
The buttons of right-side for the standalone emulator for G&W are wrongly mapped, the pair A-Y is set as B-X and us difficult using.

Rezzy-dev, 2023-01-10:
Yep... the standalone G&W emulator still needs work... The keymapping was left the PC default, and there's no way to change it without recompiling the source.

salvacam, 2023-01-10:
I have already changed the mapping of the A,B,X and Y buttons

You have to change this file in the /emus/game&watch folder of the main partition
(rename the file to lr, github doesn't allow to upload file without extension)

[lr.txt](https://github.com/TriForceX/MiyooCFW/files/10386434/lr.txt)

When I have time I upload it to a repository and link it here

eknives, 2023-01-10:
It works, it's perfect. Thank you very much.

Rezzy-dev, 2023-01-11:
I'm happy to stand corrected! :D \*^_^\* Thanks @salvacam. 🤗

salvacam, 2023-01-15:
> @salvacam Thank you for looking into this. I updated the emulator on a V90 and the Select button is still exiting the app instead of cycling through the buttons, I am unable to start most games. Would you be able to remap the exit to a different button, perhaps the Fn key? Thank You.

Check this file 
[lr.txt](https://github.com/TriForceX/MiyooCFW/files/10420058/lr.txt)
I change the exit button, now to exit use the button Reset

You rewrite this file in the /emus/game&watch folder of the main partition
(rename the file to lr, github doesn't allow to upload file without extension)

salvacam, 2023-01-15:
Upload code with changes and release in this repository
https://github.com/salvacam/gw_miyoo/

an-eel, 2023-01-15:
Yessss, it's working great. The select button is cycling through the options and I can use Start to push those button. The exit is being triggered by pressing the Fn button. Thank you so much for the work you are putting in, it is really appreciated by people like me.

Rezzy-dev, 2023-01-16:
@salvacam Added GW code/repository to main page (readme):

https://github.com/TriForceX/MiyooCFW

Closing this issue as fixed/completed.

closed 2023-01-16 by Rezzy-dev