TriForceX/MiyooCFW #458: PowKiddy Q90 Plus button unmapped

eknives, 2023-01-18:
Hi, Is possible to map the Plus button (right side)? It can be useful for some emulators, Q90 hasn't buttons L2/R2.

Thx.

szymor, 2023-01-18:
Not sure, but as far as I remember it is a dummy button. Somebody else needs to confirm, though.

TriForceX, 2023-01-18:
Yes is possible, we confirmed it with @Apaczer few weeks ago, the only one first we tested was using an outdated/different tool...

Theres actually a key combo to simulate L2/R2, we can consider it, but needs to be something not very important due that button is only present on Q90 and Q20.

Thanks for your suggestion!

eknives, 2023-01-18:
Thanks. It is only needed for psx emulation. Sometimes I switch select/start with L2/R2 but I lost select/start in game. Then I usually set menu (emulator) and start button in minus button because is not a matter pause game when I need to open emulator options and if I need press start the button stills available.

Rezzy-dev, 2023-01-19:
> ...it [...] needs to be something not very important due that button is only present on Q90 and Q20.

Actually, no, if it's available, let's map it and make it settable/usable in emulators! An extra button is an extra button! Why exclude it?

eknives, 2023-01-19:
> 

It is cool that we can map an extra button, but I still thinking that select+R=R2, select+L=L2, start+R=start and start+L=select is a better choice when you are using pcsx rearmed standalone emulator. This is the really improvement for this cfw

Rezzy-dev, 2023-01-19:
> yea the only reference we had was my console, nobody else wanted to test it on another q90, my mistake was that the console i used was an early version (it didn't have the powkiddy label and the printed model was LD500) and ive used a wrong iotester version, so when i tried it on another of my q90s with the proper tool i noticed that it did work.

Oh, I see! 😮  But I've been here all along, and developing on a new PowKiddy Q90! 😅  And I'm more than happy to help get this working.

When you say it works, do you mean that the button physically works, or that it can already be configured/mapped in the system?

I've tried to run `xev` and `showkey` to test the input, but neither are installed on our build of Miyoo Linux.

TriForceX, 2023-01-19:
> Oh, I see! 😮 But I've been here all along, and developing on a new PowKiddy Q90! 😅 And I'm more than happy to help get this working.
>
> When you say it works, do you mean that the button physically [...]

The test we did few weeks ago i got PAGE DOWN on the + "fake" button
[image]
then we did some tests to change it or testing new hotkeys, cannot remember right now, i summon @Apaczer to tell you the latest status

Apaczer, 2023-01-19:
I've been summoned! Please state your wish mortals :smiley: (sorry, had to ;p)

Yes, we've been working with @TriForceX to make Right function ("+") somewhat usable. Turns out it gave input signal long since L2/R2 buttons on V90 have been supported by @flabbergast, hence PAGE_DOWN output on _I/O tester_ when "+" button is pressed which corresponds to R2 on V90.

The right upper button on Q90 and Q20 mini (let's call it Rfunction) are connected to the same GPIO, so behave exactly the same.

Firstly with Q20 mini we created separate variant for keyboard layout to swap Rfunction/Lfunction buttons & include new hotkey combo for these devices, by request of users:
- map Rfucntion to "RESET" (thus hotkey activator)
- map Lfunction to "L3"
- activate "R3" with hotkey combo thus RESET+L3

``Starting from 2.0 CFW the "L2/R2" simulation will be activated with RESET+L1 & RESET+R1 combo on all devices (beside V90 which has reserved those for "L3/R3").``

Secondly I tried to add Q90 variant which will consider swapped location of physical SELECT/START buttons comparingly to Q20 mini, and thought it might be worth a shot and do smth like this:
- leave Lfunction button so it is "RESET" (thus hotkey activator)
- move brightness/volume hotkeys from _D-PAD_ to _FaceButtons_
- map Rfunction to "R3"
- activate "L3" with hotkey combo thus RESET+R3

However finally we decided to use the same variant for Q90 and Q20, as using face buttons for brigthness/volume control may feel unintuitive for some users.
___
**If u guys feel that anyhing should be changed for Q90, please let us know (we value every thoughtful vote). Firstly check on your device these images (changing naming in console.cfg accordingly):**
- official ``q20`` variant from [MiyooCFW/sdcard CI build](https://github.com/MiyooCFW/sdcard/actions/runs/3962511528) 
and
- unofficial ``q90_v2`` variant from [offbranch fork](https://github.com/Apaczer/sdcard/actions/runs/3744059146)

Rezzy-dev, 2023-01-20:
Thanks @TriForceX, @Apaczer. I don't have a problem at all with the way the current 1.3.3 firmware works on the Q90, actually. The only thing missing in terms of hardware functioning has been the Plus button.

It is great to know that the button actually fully works in the firmware! 😄  ❤️  Thank you for this discovery, and the really helpful screenshot, @TriForceX. I will try wiring the button in in the configs, and see how I go (will let you know).

I presume (I think correctly) that 281 is the SDL keycode for it.

Rezzy-dev, 2023-01-20:
Ah, I must have misunderstood. It's not implemented into the system yet, right? Because SDL doesn't seem to be recognising it on DOSBox.

Apaczer, 2023-01-20:
Yes, u must first flash one of alpha *.img builds I have linked in prv
reply already.

Rezzy-dev, 2023-01-20:
Yep, I wasn't sure as to the timeframe this falls in. I mistook it for before 1.3.3 was released:

> Turns out it gave input signal long since L2/R2 buttons on V90 have been supported by @flabbergast, hence PAGE_DOWN output on I/O tester when "+" button is pressed which corresponds to R2 on V90.

I will try the new, WIP build when I get a chance to.

(If this works, which I'm presume it will, and if the new WIP build is stable, I'm just not keen on having to migrate everything over from the previous, stable build. Not sure I'll have the time to at the moment. So that cuts my enthusiasm a bit. The new build is essentially a new OS release, and I'm not sure I'm ready to upgrade. Quite happy with my old setup. But that doesn't mean I can't test the new build on a new SD card.)

eknives, 2023-01-20:
I have tested in my PowKiddy Q90 this one official q20 variant from [MiyooCFW/sdcard CI build](https://github.com/MiyooCFW/sdcard/actions/runs/3962511528)

and minus button is ok, plus button is page down, minus button + L button is page up, but minus button + R button is page down too.

How can I avoid BG Style PowKiddy logo? I like it, but is better if is optional.

Thanks.

Rezzy-dev, 2023-01-21:
> How can I avoid GB Style PowKiddy logo? I like it, but is better if is optional.

@eknives Very simple: remove the "boot-logo" file on the "main" partition. It contains that animation/logo. If it's not there, the boot process will skip it.

eknives, 2023-01-21:
> > How can I avoid GB Style PowKiddy logo? I like it, but is better if is optional.
> 
> @eknives Very simple: remove the "boot-logo" file on the root of the "main" partition. It contains that animation/logo. If it's not there, the boot process will skip it. It _is_ optional.

It isn't there in new CFW 2.0.

Apaczer, 2023-01-21:
eknives, if u changed variant in console.cfg u must perform FIRSTBOOT
procedure again, thus change in /boot file ``firstboot.done`` to
``firstboot``, otherwise new key bindings won't apply.

eknives, 2023-01-21:
@Apaczer I didn't change anything and everything seems to be fine, I only asked how to eliminate classic GB boot animation.

Apaczer, 2023-01-21:
Change in /boot/console.cfg to CONSOLE_VARIANT=q20 and then perform
firstboot procedure.

eknives, 2023-01-21:
@Apaczer I did it just now, Plus is Rshift.

Apaczer, 2023-03-11:
Updated inputs on latest build and overview in: <https://github.com/TriForceX/MiyooCFW/wiki/Miyoo-CFW#console-hotkeys>

closed 2023-03-11 by Apaczer