TriForceX/MiyooCFW #459: PowKiddyQ90/Q20: Combo buttons for L2/R2 in PCSX reARMed standalone

eknives, 2023-01-19:
Hi, Would it be possible to modify or create a proper config file for PCSX rearmed standalone (and Retroarch) with button combinations to get L2/R2 working on PowKiddy Q90?

I think a good solution would be Select+L=L2 , Select+R=R2 , Start+L = Select , Start+R= Start.

Thx.

Rezzy-dev, 2023-01-19:
There's a hidden folder inside the emus/pcsx_rearmed/ directory, called ".pcsx". Inside it you'll find "pcsx.cfg". This is the emulator's configuration file, that contains the button mapping setup. You can change the keys assigned to each PS controller button here, but I am not sure that PCSX-R supports combination keys to set a button (what you are trying to do with Select + L). You could play around with it and see. Just make sure you back up your original "pcsx.cfg" file before you do.

The Select button on Miyoo corresponds to the ESC key, and the Start button to the Enter key. L is Tab, and R is Backspace.

eknives, 2023-01-19:
Thanks but ni ways. I also tried with "and", & and others combinations.

Rezzy-dev, 2023-01-20:
Yeah, I just checked in the [source code](https://github.com/steward-fu/miyoo/releases/download/v1.2/pcsx_rearmed.7z). Such key combinations are not supported by the emulator (haven't been programmed into it). You will need to assign L2 and R2 to single buttons.

Rezzy-dev, 2023-01-20:
Note that you should be able to simulate an L2 and R3 button press using the Reset + L / R combination. This is a system-wide setting, and it should work in the emulator, too.

Rezzy-dev, 2023-01-20:
It looks like you'll have an extra button to use! Just testing it now, but it looks like the Plus button on the Q90/Q20 is actually fully functional, and corresponds to the Page Down keyboard key. You could 'wire' it to either L2 or R2 for the emulator using the config settings above:

```
bind page down = player1 L2
```

or

```
bind pagedown = player1 L2
```

Apaczer, 2023-01-20:
The "Simulated L2 & R2" can be accesed since 1.3.1 CFW with following combo:
L2 = RESET+TB (on Bittboy)
R2 = RESET+TA (on Bittboy)
L2 = RESET+X (on others)
R2 = RESET+A (on others)

Our wiki description may need correction as it is referring to next 2.0
release (we should add that info as side note)

eknives, 2023-01-21:
In CFW 1.3.3 I use Select/Start like L2/R2 in emulator and the combos Minus+X/Minus+A like Select/Start in emulator.

In new CFW 2.0 I use Select/Start like L2/R2 in emulator, Reset+L to open menu emulator and Minus/Plus like Select/Start.

Rezzy-dev, 2023-01-21:
Thanks for that, @Apaczer. We really should update the documentation to reflect this -- most users are still on 1.3.3. The documentation for 1.3.3 and for 2.0 should be split, as they are two entirely different OS releases, with different setups.

What SDL keycodes (or keyboard keys) do L2 and R2 correspond to?

Apaczer, 2023-01-21:
Rezzy, yes old/new CFW differ quiet much at current state:

1.3.3 CFW:
RALT - L2
RSHIFT - R2

2.0 CFW:
PAGEUP - L2
PDOWN - R2
RALT - L3
RSHIFT - R3

Apaczer, 2023-03-11:
The hotkey related wiki section has been updated: <https://github.com/TriForceX/MiyooCFW/wiki/Miyoo-CFW#console-hotkeys>.

If you're on 1.3.3 CFW look at Bittboy column (it applies through out all consoles there)