TriForceX/MiyooCFW #486: FCEUX doesn't play roms on my V90

Rinka-Sei, 2023-04-07:
I have followed the steps on installing the cfw on my sd card, but for some reason FCEUX doesn't work. I tried to play a rom but the screen turns black for a brief moment like gpsp does when there's no BIOS file then it sends me back to the list of my roms

I tried using a BIOS file called "disksys.rom" thinking that the emulator needs a bios file but that didn't solve the problem. Anybody knows this problem and how to solve it? I got my nes roms from no-intro and they seem clean so I don't think it's the roms.

Apaczer, 2023-04-08:
1. Which version of CFW?
2. What device are you using?
3. Show us screen with logs after app crashing (Enable in gmenu2x settings -> OUTPUT LOGS: ON)

Rinka-Sei, 2023-04-08:
[image]

1. I use cfw ver 1.3.3 for powkiddy q90/v90
2. Powkiddy V90 Black
3. It shows "an error occurred..." On the log file

Rinka-Sei, 2023-04-08:
When I use my old sd with the stock cfw (nxhope) that came with the V90, fceux works just fine. So i tried copying the fceux folder from main root and from /emu from my stock to my cfw sd and that didn't work as well. I also reflashed the cfw twice and redownloading the cfw per reflash, that didn't work as expected.