TriForceX/MiyooCFW #499: An option to hide the image folder of a rom folder, but still show the rest of the folders

akg83, 2023-06-02:
It could be a nice quality-of-life improvement, if we could hide the boxart image folder that is found in the rom folder of a specific emulator, but without losing the ability to use folders inside of it. so basically to have the option "Hide Image folder" for each roms folder, which will hide the Img/Image folder found inside of it, but still show the other folders.

This can be achieved by using a specific name for said boxart image folder, so in order for this to work the image folder will have to be always "Img" (three letters, capital "I"),

this option should be a toggle that is available under an "edit" menu when editing a rom folder options.

[image]

salvacam, 2023-06-02:
If what you want is that it always hides the images folder, no development is necessary. Rename the folder by adding a dot at the beginning of the name (with this your folder will be hidden) 
[image]

and update the emulator file in the gmenu folder the selectorscreens line
[image]

akg83, 2023-06-02:
> If what you want is that it always hides the images folder, no development is necessary. Rename the folder by adding a dot at the beginning of the name (with this your folder will be hidden) [image]
> 
> and update the emulator file in the gmenu folder the selectorscreens line [image]

adding a dot prefix to a folder name is making the folder a "hidden folder", and how will I be able to later select this path so that the boxart get loaded?  I assume that the file navigator of the CFW will treat this folder as hidden as well ?

Apaczer, 2023-06-04:
There could be "Show hidden folders" option in general gmenu2x settings  which would make more sense than just making whole effort to hide one boxart folder :file_folder: . Alternatively I was thinking of adding "Developer mode" for accessing also read-only stuff like root directory or ``/boot``