TriForceX/MiyooCFW #506: Any steps to build the cfw for F1e200

narioinc, 2023-09-16:
Hi All

I just got a SUP m3 but it seems to run on the new F1e200 chipset. Now, i did read through the wiki that the support for an SUP m3 with a F1e200 is not there.
I just wanted to check with the community on what are the general steps if i would like to build the CFW for f1e200 ?? I have decent enough knowledge working on linux and just some high level guidance would helm me get  started on the journey to support the f1e200 chipset.

szymor, 2023-09-16:
The chip is simply not supported. It is not possible to run CFW on it without serious development effort.

narioinc, 2023-09-16:
Thanks for the quick reply ! 
This issue can be closed.

closed 2023-09-16 by narioinc