TriForceX/MiyooCFW #523: [Question] Building from source

EchedeyLR, 2023-10-23:
I have been reading the build guide you host at https://github.com/TriForceX/MiyooCFW/wiki/Build-Source but there are several faulty things:

1. `buildroot` seems to include now the toolchain, does this affect to the guide described here or the building process?
2. The `uboot` section seems faulty to where to flash it. If a person wants to use it in the sdcard, to which partition should they flash it or is indeed saved at MBR or similar?
3. The building kernel section, what is the difference between hardware revisions to know which to choose? or I must choose both? (I guess is only one given the next steps) Does it depend on the console?, the microSD?
4. Is firmware (real one, not this OS) and/or drivers built/added with this process or I must gather it from somewhere? I don't know the difference between the releases for each console.

szymor, 2023-10-23:
Ad 1. You use buildroot to build the toolchain. The toolchain is not a part of buildroot.
Ad 2. There is no mistake - you do not flash U-Boot to a partition but to a specific SD card sector/range of sectors.
Ad 3. Hardware revision is specified by the console you own. Different revisions have slight differences in PCB layout and components.
Ad 4. This is real firmware. It includes u-boot, kernel, drivers and some user apps.

EchedeyLR, 2023-10-23:
I don't understand the first reply, but I do with the rest.

Where is the toolchain hosted, build then? The link referenced references to the old repo which is now deprecated in favor of buildroot and it affirms that the code is there now or so.

Apaczer, 2023-10-23:
Our WIKI refers to the build process of the NxHope (first custom firmware for Miyoo) written by @steward-fu. The method to make MiyooCFW from scratch has changed since that time, which derived from former - sorry but we didn't update it since then much.

What you really need TBH is only the [buildroot](https://github.com/miyoocfw/buildroot/) (see README there). If you want to make modifications to different packages just add *.patch in correspoding `package/<package_name/` directory. You can also run `make <package_name>-extract` to fetch extracted tarbal src, which you will find then in `output/build/<package_name>-<version>` dir to apply changes directly to source. More info in buildroot's [manual](https://github.com/MiyooCFW/buildroot#buildroot-2022029-docs-httpsgithubcommiyoocfwbuildrootblobmasterdocsmanualmanualpdf).

EchedeyLR, 2023-10-23:
I guess I will wait for the process to be documented. The buildroot manual is too big and includes information that does not seem to be needed for the current use case and are more general and modular like it I wanted to tweak.

My current objective is getting a reproducible build of the firmware as far as you released it or you would release it so I can automate some parts and then use it in my console.

closed 2024-10-16 by Apaczer