TriForceX/MiyooCFW #535: TVMODE from options.cfg not working

DaemonSlayerZero, 2023-12-31:
Some user have enabled TVOut without connected to a tv, so the only current way i know is removing tvout file from main partition, but i noticed theres also TVMODE in options.cfg, is this supposed to work? because ive tried to swap between 0 and 1 many times without results, even when TVOut was enabled from console it didnt changed in the cfg file.

[image]

Apaczer, 2023-12-31:
See: https://github.com/MiyooCFW/sdcard/blob/35f11921df5b7257f58dfee02643d4b48f282077/README.md?plain=1#L117

To disable TVout you can hold "SELECT" while in gmenu2x. This is not ideal, but we might think of other way for eg at startup: "TVout enabled. Continue? Reverting in 10s"

closed 2023-12-31 by Apaczer

MayanKoyote, 2024-01-01:
> To disable TVout you can hold "SELECT" while in gmenu2x. This is not ideal, but we might think of other way for eg at startup: "TVout enabled. Continue? Reverting in 10s"

blind hold of "SELECT" is fine.
but extra reminder message "'SELECT' (hold): Disconnect TV-out" at startup won't be amiss.
https://github.com/TriForceX/MiyooCFW/issues/145#issuecomment-1872261153