TriForceX/MiyooCFW #538: Control pads don't seem to work with the USB host.

IDOCARE1, 2023-12-31:
I've tried an 8bitdo pro and another one, but none seem to work. I have tried using a usb A to USB A/B split to power it off a power bank, and just plugging them in directly.

tiopex, 2024-01-02:
Hi, you have to add mapping for your gamepad to file `joymap.map` https://github.com/MiyooCFW/buildroot/blob/master/board/miyoo/main/joymap.map

`vendor`  and `product` id you can get from `lsusb` command

button mapping from your gamepad you can get by running `input_info.sh` script https://github.com/MiyooCFW/buildroot/blob/master/board/miyoo/main/apps/joymap/input_info.sh

After all, you have to run `joymap.sh` https://github.com/MiyooCFW/buildroot/blob/master/board/miyoo/main/apps/joymap/joymap.sh

IDOCARE1, 2024-01-02:
Oh....yeah, I did none of that.

So, just so I get the process, is this something I can do after plugging the pad into the V90, or do I need to edit the files via a laptop the first time?

tiopex, 2024-01-02:
you need to connect pad to v90 and run host mode, to get output from `lsusb` and `input_info.sh`

IDOCARE1, 2024-01-02:
Hmm, I got the info from lsusb in the simple terminal, but when I execute input_info.sh from the commander app, it just takes me back to the main menu.

tiopex, 2024-01-02:
`input_info.sh` creates `input.log` file with gamepad buttons id in main partition

IDOCARE1, 2024-01-02:
Oh, the input.log file in the main partition has no data in it.

tiopex, 2024-01-02:
So the device was not detected. You can check if the device is detected by plug-in/plug-out the device and run the `lsusb` command. After plug-in, the new entry should appear in the output of the `lsusb`. 

Please note that the device must be externally powered and host mode must be turned on. You can check if detection works with any USB device e.g. mouse or keyboard.

IDOCARE1, 2024-01-02:
Oh, yeah, you are right, the lsusb was just reporting what it normally would with nothing plugged in.

Thanks for your help, but I think this one is beyond my limited tech knowledge!

closed 2024-01-02 by IDOCARE1