TriForceX/MiyooCFW #539: TV out slowdown

IDOCARE1, 2024-01-02:
RA cores seem to run even slower using TV out. You can tell by listening to the stuttering sound.

tiopex, 2024-01-03:
Yes, performance on TV out is slower due to use 32bpp mode instead 16bpp. Unfortunately, f1c100 HW scaler can works only 32bpp mode, 16bppmode is not supported, and nothing we can do with this.

IDOCARE1, 2024-01-03:
That's cool, thanks.

Still works fine for JRPG's without the sound on, you don't notice the slight stutter :D

actraiser30, 2024-01-27:
Hi is 24bit supported on tv out??

And another thing is I used the less compatible image and with tv out enabled anything I was running eg: gb/nes was not working and returning to the gui,is this a known issue???

Apaczer, 2024-01-28:
>  less compatible image and with tv out enabled anything I was running eg: gb/nes was not working

looks like it is a GMenu2X issue on `musl` build, thanks for reporting

actraiser30, 2025-04-15:
is this fixed? or going to be fixed?
cant really play anything with tv-out enabled
thx