TriForceX/MiyooCFW #546: Any chance for bilinear image interpolation?

Xtremis77, 2024-01-22:
I was looking at the image quality on devices from more or less around the same time / similar specs, and the LDK Game, and the Model S, are able to do bilinear image interpolation, and smooth filtering, respectively. I really like the blurry, smooth look on my smaller handhelds :)

I've noticed that on the 2.0 beta, there is no bilinear option in RA (only semi-linear in image interpolation). Is there any chance for having bilinear filtering in RA with version 2.0? I don't know how taxing this might be to performance, but seeing that the Model S (which as the same specs as the V90, although with a smaller screen) has a smoothing option, I think it might be worth a shot at asking :)

Thanks for your incredible work!

Apaczer, 2024-01-26:
The LDK is an Ingenic device which uses HW scalling by onboard GPU (not preset on our devices), for that they have separate video driver on RA (sdl_dingux_gfx). The other spectrum is Trimui Model S, which rely on picoarch a minimal frontend for libretro cores (think of it as completely different app able to load RetroArch cores) - original author decide to go with custom [pixel interpolation method](https://git.crowdedwood.com/picoarch/commit/?id=8ad26356f5e92bd396e58290217da55858345a4e).

Official Miyoo's RetroArch build use `sdl_rs90_gfx` driver for max performace, thus other [pixel mixing](https://github.com/libretro/RetroArch/commit/218286aa99b39a766e83d41f119f03df67bcf367). There is an option to load standard `sdl` driver with bilinear filtering, but it is waaaay slower that semi-linear. For now I suggest for you to try different "Video Filters". maybe they will suit you.

Xtremis77, 2024-01-26:
Understood, thanks, @Apaczer ! :) I'll explore the filters :)

Apaczer, 2024-01-26:
There is a possibility we will add extra image interpolation with more blur (some day), let us leave this issue open until then! Will require some work 🏋️