TriForceX/MiyooCFW #554: [BUG]: Slow Performance in GBA Emulation

Timman6866, 2024-02-13:
### Version

2.0.0 (BETA) pre-release

### Version type

uClibc (compatible), musl (experimental)

### Issue description

Emulation slowdowns exist for at least the gpSP emulator. The play speed is not consistent. Can be reproduced easily by loading a game with a clock counter. I use Tony Hawk Pro Skater 2.

### Steps to reproduce

Turn on the device. Load Tony Hawk Pro Skater 2 through RetroArch or core. Play a level and experience the slowdown by observing the running game clock.

### What device are you using?

PowKiddy V90

### Issues' terms

- [X] I did not find any existing open/closed issue of same matter

Timman6866, 2024-02-15:
This issue is specifically related to the RetroArch emulator pack. Loading the gpSP emulator as stand-alone resolved the performance issue.

EchedelleLR, 2025-08-14:
@Timman6866 how did you load the emulator standalone?