TriForceX/MiyooCFW #555: [BUG]: Volume maximum is much quieter on CFW 2.0.0 vs 1.3.3

Timman6866, 2024-02-13:
### Version

2.0.0 (BETA) pre-release

### Version type

musl (experimental)

### Issue description

Opening the same GBA ROM when running 1.3.3 is much louder than when running on CFW 2.0.0. It's obvious when playing two devices next to each other. However, I did flash both versions on the same device to rule out hardware.

### Steps to reproduce

Run a GBA ROM and turn the volume up to max. Observe that the sound is quieter than on CFW 1.3.3

### What device are you using?

PowKiddy V90

### Issues' terms

- [X] I did not find any existing open/closed issue of same matter

TriForceX, 2024-02-14:
Have you tried the same emularors on both versions? because 1.3.3 came with an old version of GPSP... also 2.0 came with retroarch also, so there are many ways to run GBA games so can you confirm this using exactly the same emus on both versions? just to be sure if the problem is the emulator, core or the cfw.

Also you said you used 2 devices, have you swapped the cfws on both? thats needed to confirm if is a device issue too.

Timman6866, 2024-02-14:
Thanks for the quick reply. It's going to take me a little time to understand the process of matching emulator versions between the two firmware. I did rule out the device by booting both 1.3.3 and 2.0.0 on the same v90 and experienced the difference in volume.

Timman6866, 2024-02-15:
The issue was a combination of audio stutter from Retroarch cores and the default global volume being 70 on 2.0.0 vs 90 on 1.3.3.

closed 2024-02-15 by Timman6866