TriForceX/MiyooCFW #567: [FEATURE-REQ]:  Add 1.6 scale and less sharpness to gambatte in retroarch

jeffersonmsb, 2024-03-16:
## 💡 Feature request

Hello everything is fine? Add 1.6 scale and less sharpness to gambatte in retroarch. 

How can I contribute to this? What would be the way? I have some experience with code.

## Motivation (behind your idea)

There is currently no such option in retroarch core, but it is available in gambatte outside of retroarch

salvacam, 2024-03-16:
For what you are asking for, I think the best thing to do is to create a retroarch filter.
You would have to create the filter and make a PR to the Retroarch repository
https://github.com/libretro/RetroArch

For example, in this PR a new filter is added
https://github.com/libretro/RetroArch/pull/13954

jeffersonmsb, 2024-03-17:
Yes, yes, that is the best way. I should have also posted it on retroarch but I only remembered about it later.

I started reading code. gambatte-dms has this upscaler and this is not in the retroarch because it is in the SDL part of gambatte dms. I'm reading the PR you mentioned to understand how to make this filter. It would be an upscaler from 160x144 to 320x260.

The Upscale_240x160-320x240 of this PR is wonderful, really good. Much better than options outside of retroarch.

Some filters don't work on 8-bit cores, I don't understand that well yet.

jeffbzrr, 2024-11-30:
I am 90% developed from this filter. I opened a thread in discussions, about what I should do with it when the filter is ready.

jeffbzrr, 2024-12-07:
PR performed.
https://github.com/libretro/RetroArch/pull/17240

commit 1b55fba7 (2024-03-23): README: add `pixel-reader` epub viewer app

closes #567
closed 2024-12-07 by jeffbzrr