TriForceX/MiyooCFW #570: [BUG]: System times out on long USB transfers in new MTD mode

matrxbox, 2024-03-25:
### Version

2.0.0 (BETAv2) pre-release

### Version type

musl (experimental)

### Version hash

cfw=67298a0

### Issue description

The settings for system suspend and power off time out can either be completely disabled or set to as much as 300 seconds (5 minutes).  This does not allow sufficient time for large file trasnsfers over USB in MTD mode, and if you do not manually keep the system active or completely disable the timeouts, the system will suspend and the transfer will break.

Expexcted behavior is that the system recognises it is in usb gadget mode (MTD for me, but others should likely work the same) and either stay active while in the mode, or have a third configuration option to keep it active when connected.

### Steps to reproduce

1. turn on the device
2. go to settings
3. set suspend timeout to maximum 300 seconds
4. enable usb mtd
5. connect to computer via usb
6. copy a large folder over
7. ait and watch the copy lock up and the system suspend 

### What device are you using?

PowKiddy V90

### Issues' terms

- [X] I did not find any existing open/closed issue of same matter

Zach1812, 2024-03-30:
This happened to me yesterday when transferring ps1 roms to my Powkiddy Q90. On the old 1.33 software, the transfer was faster due to sd card to computer, but MTD transfer was slow.

Apaczer, 2024-10-28:
fixed via: https://github.com/MiyooCFW/gmenu2x/commit/8509f727cbd977b13fd09157328f164fa3bceade

closed 2024-10-28 by Apaczer