TriForceX/MiyooCFW #571: [BUG]: Q90 USB Mac mount not working

Zach1812, 2024-03-29:
### Version

2.0.0 (BETAv2) pre-release

### Version type

uClibc (compatible)

### Version hash

67298a0

### Issue description

When trying to mount Q90 on a mac, nothing shows up (With known good Data/Charge cable). When plugging in the sd card, the boot partition does show up, but not the main. I have a 90% charge (hard to have it show 100% these days) and have followed the directions to install the uClibc version of the Beta-2. I am using a PNY 64GB sd card that worked with this system quite well before and it 'almost' brand new (8 months).

### Steps to reproduce

1. Turn Powkiddy Q90 on and wait for boot. 
2. When main screen appears, connect known good data/charge cable from Q90 to Mac (13.6.4)
3. When USB does not show up select USB MTD.
4. USB MTD loads then fails back to main screen.

### What device are you using?

PowKiddy Q90

### Relevant log outputs

```shell
CONSOLE_PARAMETERS=lowcurrent=1
CONSOLE_VARIANT=q90
CONSOLE_VIDEO=st7789sfb.ko
FORCE_VERSION=q90
arch=arm
baudrate=115200
board=sunxi
board_name=sunxi
boot_a_script=load ${devtype} ${devnum}:${distro_bootpart} ${scriptaddr} ${prefix}${script}; source ${scriptaddr}
boot_device=mmc0
boot_extlinux=sysboot ${devtype} ${devnum}:${distro_bootpart} any ${scriptaddr} ${prefix}${boot_syslinux_conf}
boot_net_usb_start=usb start
boot_prefixes=/ /boot/
boot_script_dhcp=boot.scr.uimg
boot_scripts=boot.scr.uimg boot.scr
boot_syslinux_conf=extlinux/extlinux.conf
boot_targets=fel mmc_auto usb0 
bootargs=console=tty0 console=ttyS1,115200 panic=5 rootwait quiet lpj=1015808 root=/dev/mmcblk0p2 ro
bootcmd=env export -t 0x81000000; fatwrite mmc 0:1 0x81000000 uEnv.txt ${filesize}; run distro_bootcmd;
bootcmd_args=setenv bootargs ${bootargs} miyoo_kbd.miyoo_ver=6 miyoo_kbd.miyoo_layout=1 miyoo.miyoo_snd=1 pwm-suniv.motor_ver=2 miyoo-video.driver='st7789sfb+lowcurrent=1'
bootcmd_fel=if test -n ${fel_booted} && test -n ${fel_scriptaddr}; then echo '(FEL boot)'; source ${fel_scriptaddr}; fi
bootcmd_mmc0=devnum=0; run mmc_boot
bootcmd_mmc1=devnum=1; run mmc_boot
bootcmd_mmc_auto=if test ${mmc_bootdev} -eq 1; then run bootcmd_mmc1; run bootcmd_mmc0; elif test ${mmc_bootdev} -eq 0; then run bootcmd_mmc0; run bootcmd_mmc1; fi
bootcmd_usb0=devnum=0; run usb_boot
bootdelay=0
bootm_size=0x1700000
console=ttyS0,115200
cpu=arm926ejs
dfu_alt_info_ram=kernel ram 0x80500000 0x1000000;fdt ram 0x80C00000 0x100000;ramdisk ram 0x80D50000 0x4000000
distro_bootcmd=for target in ${boot_targets}; do run bootcmd_${target}; done
fdt_addr_r=0x80C00000
fdtcontroladdr=80e57890
fdtfile=suniv-f1c100s-generic.dtb
fileaddr=81000000
filesize=14
kernel_addr_r=0x80500000
mmc_boot=if mmc dev ${devnum}; then devtype=mmc; run scan_dev_for_boot_part; fi
mmc_bootdev=0
partitions=name=loader1,start=8k,size=32k,uuid=${uuid_gpt_loader1};name=loader2,size=984k,uuid=${uuid_gpt_loader2};name=esp,size=128M,bootable,uuid=${uuid_gpt_esp};name=system,size=-,uuid=${uuid_gpt_system};
preboot=usb start
pxefile_addr_r=0x80D00000
ramdisk_addr_r=0x80D50000
scan_dev_for_boot=echo Scanning ${devtype} ${devnum}:${distro_bootpart}...; for prefix in ${boot_prefixes}; do run scan_dev_for_extlinux; run scan_dev_for_scripts; done;
scan_dev_for_boot_part=part list ${devtype} ${devnum} -bootable devplist; env exists devplist || setenv devplist 1; for distro_bootpart in ${devplist}; do if fstype ${devtype} ${devnum}:${distro_bootpart} bootfstype; then run scan_dev_for_boot; fi; done; setenv devplist
scan_dev_for_extlinux=if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${boot_syslinux_conf}; then echo Found ${prefix}${boot_syslinux_conf}; run boot_extlinux; echo SCRIPT FAILED: continuing...; fi
scan_dev_for_scripts=for script in ${boot_scripts}; do if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${script}; then echo Found U-Boot script ${prefix}${script}; run boot_a_script; echo SCRIPT FAILED: continuing...; fi; done
scriptaddr=0x80C50000
silent=1
soc=sunxi
splashfile=miyoo-splash.bmp
splashimage=0x80000000
splashsource=mmc_fs
stderr=serial,vga
stdin=serial,usbkbd
stdout=serial,vga
usb_boot=usb start; if usb dev ${devnum}; then devtype=usb; run scan_dev_for_boot_part; fi
uuid_gpt_esp=c12a7328-f81f-11d2-ba4b-00a0c93ec93b
uuid_gpt_system=69dad710-2ce4-4e3c-b16c-21a1d49abed3
```

### Issues' terms

- [X] I did not find any existing open/closed issue of same matter

Zach1812, 2024-03-29:
Just re-ran setup with a different sd card and used the musl (Beta-2) version. Still no joy on Mac.

'''
CONSOLE_PARAMETERS=lowcurrent=1
CONSOLE_VARIANT=q90
CONSOLE_VIDEO=st7789sfb.ko
FORCE_VERSION=q90
arch=arm
baudrate=115200
board=sunxi
board_name=sunxi
boot_a_script=load ${devtype} ${devnum}:${distro_bootpart} ${scriptaddr} ${prefix}${script}; source ${scriptaddr}
boot_device=mmc0
boot_extlinux=sysboot ${devtype} ${devnum}:${distro_bootpart} any ${scriptaddr} ${prefix}${boot_syslinux_conf}
boot_net_usb_start=usb start
boot_prefixes=/ /boot/
boot_script_dhcp=boot.scr.uimg
boot_scripts=boot.scr.uimg boot.scr
boot_syslinux_conf=extlinux/extlinux.conf
boot_targets=fel mmc_auto usb0 
bootargs=console=tty0 console=ttyS1,115200 panic=5 rootwait quiet lpj=1015808 root=/dev/mmcblk0p2 ro
bootcmd=env export -t 0x81000000; fatwrite mmc 0:1 0x81000000 uEnv.txt ${filesize}; run distro_bootcmd;
bootcmd_args=setenv bootargs ${bootargs} miyoo_kbd.miyoo_ver=6 miyoo_kbd.miyoo_layout=1 miyoo.miyoo_snd=1 pwm-suniv.motor_ver=2 miyoo-video.driver='st7789sfb+lowcurrent=1'
bootcmd_fel=if test -n ${fel_booted} && test -n ${fel_scriptaddr}; then echo '(FEL boot)'; source ${fel_scriptaddr}; fi
bootcmd_mmc0=devnum=0; run mmc_boot
bootcmd_mmc1=devnum=1; run mmc_boot
bootcmd_mmc_auto=if test ${mmc_bootdev} -eq 1; then run bootcmd_mmc1; run bootcmd_mmc0; elif test ${mmc_bootdev} -eq 0; then run bootcmd_mmc0; run bootcmd_mmc1; fi
bootcmd_usb0=devnum=0; run usb_boot
bootdelay=0
bootm_size=0x1700000
console=ttyS0,115200
cpu=arm926ejs
dfu_alt_info_ram=kernel ram 0x80500000 0x1000000;fdt ram 0x80C00000 0x100000;ramdisk ram 0x80D50000 0x4000000
distro_bootcmd=for target in ${boot_targets}; do run bootcmd_${target}; done
fdt_addr_r=0x80C00000
fdtcontroladdr=80e57890
fdtfile=suniv-f1c100s-generic.dtb
fileaddr=81000000
filesize=14
kernel_addr_r=0x80500000
mmc_boot=if mmc dev ${devnum}; then devtype=mmc; run scan_dev_for_boot_part; fi
mmc_bootdev=0
partitions=name=loader1,start=8k,size=32k,uuid=${uuid_gpt_loader1};name=loader2,size=984k,uuid=${uuid_gpt_loader2};name=esp,size=128M,bootable,uuid=${uuid_gpt_esp};name=system,size=-,uuid=${uuid_gpt_system};
preboot=usb start
pxefile_addr_r=0x80D00000
ramdisk_addr_r=0x80D50000
scan_dev_for_boot=echo Scanning ${devtype} ${devnum}:${distro_bootpart}...; for prefix in ${boot_prefixes}; do run scan_dev_for_extlinux; run scan_dev_for_scripts; done;
scan_dev_for_boot_part=part list ${devtype} ${devnum} -bootable devplist; env exists devplist || setenv devplist 1; for distro_bootpart in ${devplist}; do if fstype ${devtype} ${devnum}:${distro_bootpart} bootfstype; then run scan_dev_for_boot; fi; done; setenv devplist
scan_dev_for_extlinux=if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${boot_syslinux_conf}; then echo Found ${prefix}${boot_syslinux_conf}; run boot_extlinux; echo SCRIPT FAILED: continuing...; fi
scan_dev_for_scripts=for script in ${boot_scripts}; do if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${script}; then echo Found U-Boot script ${prefix}${script}; run boot_a_script; echo SCRIPT FAILED: continuing...; fi; done
scriptaddr=0x80C50000
silent=1
soc=sunxi
splashfile=miyoo-splash.bmp
splashimage=0x80000000
splashsource=mmc_fs
stderr=serial,vga
stdin=serial,usbkbd
stdout=serial,vga
usb_boot=usb start; if usb dev ${devnum}; then devtype=usb; run scan_dev_for_boot_part; fi
uuid_gpt_esp=c12a7328-f81f-11d2-ba4b-00a0c93ec93b
uuid_gpt_system=69dad710-2ce4-4e3c-b16c-21a1d49abed3
'''

Zach1812, 2024-03-29:
Decided to try for the 2.0 Beta releases. First up is the uClibc version.

'''
CONSOLE_PARAMETERS=lowcurrent=1
CONSOLE_VARIANT=q90
CONSOLE_VIDEO=st7789sfb.ko
FORCE_VERSION=q90
arch=arm
baudrate=115200
board=sunxi
board_name=sunxi
boot_a_script=load ${devtype} ${devnum}:${distro_bootpart} ${scriptaddr} ${prefix}${script}; source ${scriptaddr}
boot_device=mmc0
boot_extlinux=sysboot ${devtype} ${devnum}:${distro_bootpart} any ${scriptaddr} ${prefix}${boot_syslinux_conf}
boot_net_usb_start=usb start
boot_prefixes=/ /boot/
boot_script_dhcp=boot.scr.uimg
boot_scripts=boot.scr.uimg boot.scr
boot_syslinux_conf=extlinux/extlinux.conf
boot_targets=fel mmc_auto usb0 
bootargs=console=tty0 console=ttyS1,115200 panic=5 rootwait quiet lpj=1015808 root=/dev/mmcblk0p2 ro
bootcmd=env export -t 0x81000000; fatwrite mmc 0:1 0x81000000 uEnv.txt ${filesize}; run distro_bootcmd;
bootcmd_args=setenv bootargs ${bootargs} miyoo_kbd.miyoo_ver=6 miyoo_kbd.miyoo_layout=1 miyoo.miyoo_snd=1 pwm-suniv.motor_ver=2 miyoo-video.driver='st7789sfb+lowcurrent=1'
bootcmd_fel=if test -n ${fel_booted} && test -n ${fel_scriptaddr}; then echo '(FEL boot)'; source ${fel_scriptaddr}; fi
bootcmd_mmc0=devnum=0; run mmc_boot
bootcmd_mmc1=devnum=1; run mmc_boot
bootcmd_mmc_auto=if test ${mmc_bootdev} -eq 1; then run bootcmd_mmc1; run bootcmd_mmc0; elif test ${mmc_bootdev} -eq 0; then run bootcmd_mmc0; run bootcmd_mmc1; fi
bootcmd_usb0=devnum=0; run usb_boot
bootdelay=0
bootm_size=0x1700000
console=ttyS0,115200
cpu=arm926ejs
dfu_alt_info_ram=kernel ram 0x80500000 0x1000000;fdt ram 0x80C00000 0x100000;ramdisk ram 0x80D50000 0x4000000
distro_bootcmd=for target in ${boot_targets}; do run bootcmd_${target}; done
fdt_addr_r=0x80C00000
fdtcontroladdr=80e57890
fdtfile=suniv-f1c100s-generic.dtb
fileaddr=81000000
filesize=14
kernel_addr_r=0x80500000
mmc_boot=if mmc dev ${devnum}; then devtype=mmc; run scan_dev_for_boot_part; fi
mmc_bootdev=0
partitions=name=loader1,start=8k,size=32k,uuid=${uuid_gpt_loader1};name=loader2,size=984k,uuid=${uuid_gpt_loader2};name=esp,size=128M,bootable,uuid=${uuid_gpt_esp};name=system,size=-,uuid=${uuid_gpt_system};
preboot=usb start
pxefile_addr_r=0x80D00000
ramdisk_addr_r=0x80D50000
scan_dev_for_boot=echo Scanning ${devtype} ${devnum}:${distro_bootpart}...; for prefix in ${boot_prefixes}; do run scan_dev_for_extlinux; run scan_dev_for_scripts; done;
scan_dev_for_boot_part=part list ${devtype} ${devnum} -bootable devplist; env exists devplist || setenv devplist 1; for distro_bootpart in ${devplist}; do if fstype ${devtype} ${devnum}:${distro_bootpart} bootfstype; then run scan_dev_for_boot; fi; done; setenv devplist
scan_dev_for_extlinux=if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${boot_syslinux_conf}; then echo Found ${prefix}${boot_syslinux_conf}; run boot_extlinux; echo SCRIPT FAILED: continuing...; fi
scan_dev_for_scripts=for script in ${boot_scripts}; do if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${script}; then echo Found U-Boot script ${prefix}${script}; run boot_a_script; echo SCRIPT FAILED: continuing...; fi; done
scriptaddr=0x80C50000
silent=1
soc=sunxi
splashfile=miyoo-splash.bmp
splashimage=0x80000000
splashsource=mmc_fs
stderr=serial,vga
stdin=serial,usbkbd
stdout=serial,vga
usb_boot=usb start; if usb dev ${devnum}; then devtype=usb; run scan_dev_for_boot_part; fi
uuid_gpt_esp=c12a7328-f81f-11d2-ba4b-00a0c93ec93b
uuid_gpt_system=69dad710-2ce4-4e3c-b16c-21a1d49abed3
'''

And then again with the musl version. Again with a different SD card
 '''
CONSOLE_PARAMETERS=lowcurrent=1
CONSOLE_VARIANT=q90
CONSOLE_VIDEO=st7789sfb.ko
FORCE_VERSION=q90
[... 31 lines cut ...]
filesize=14
kernel_addr_r=0x80500000
mmc_boot=if mmc dev ${devnum}; then devtype=mmc; run scan_dev_for_boot_part; fi
mmc_bootdev=0
partitions=name=loader1,start=8k,size=32k,uuid=${uuid_gpt_loader1};name=loader2,size=984k,uuid=${uuid_gpt_loader2};name=esp,size=128M,bootable,uuid=${uuid_gpt_esp};name=system,size=-,uuid=${uuid_gpt_system};
preboot=usb start
pxefile_addr_r=0x80D00000
ramdisk_addr_r=0x80D50000
scan_dev_for_boot=echo Scanning ${devtype} ${devnum}:${distro_bootpart}...; for prefix in ${boot_prefixes}; do run scan_dev_for_extlinux; run scan_dev_for_scripts; done;
scan_dev_for_boot_part=part list ${devtype} ${devnum} -bootable devplist; env exists devplist || setenv devplist 1; for distro_bootpart in ${devplist}; do if fstype ${devtype} ${devnum}:${distro_bootpart} bootfstype; then run scan_dev_for_boot; fi; done; setenv devplist
scan_dev_for_extlinux=if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${boot_syslinux_conf}; then echo Found ${prefix}${boot_syslinux_conf}; run boot_extlinux; echo SCRIPT FAILED: continuing...; fi
scan_dev_for_scripts=for script in ${boot_scripts}; do if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${script}; then echo Found U-Boot script ${prefix}${script}; run boot_a_script; echo SCRIPT FAILED: continuing...; fi; done
scriptaddr=0x80C50000
silent=1
soc=sunxi
splashfile=miyoo-splash.bmp
splashimage=0x80000000
splashsource=mmc_fs
stderr=serial,vga
stdin=serial,usbkbd
stdout=serial,vga
usb_boot=usb start; if usb dev ${devnum}; then devtype=usb; run scan_dev_for_boot_part; fi
uuid_gpt_esp=c12a7328-f81f-11d2-ba4b-00a0c93ec93b
uuid_gpt_system=69dad710-2ce4-4e3c-b16c-21a1d49abed3
'''

TriForceX, 2024-03-29:
Thanks for reporting, we doesnt have many people using OSX so this is useful. Can you try the next stuff?

For USB MTD connecting method:
https://github.com/ganeshrvel/openmtp

For direct SD card reader:
https://www.sobyte.net/post/2022-03/how-to-reading-btrfs-partition-files-under-macos/

Zach1812, 2024-03-30:
Yes. for Beta-2 the USB MTD method works. I was able to install it, and easily transfer my files to the sd card (main/roms) Kudos to that developer!

I don't have Parallel Desktop due to it's high cost and short window in which to use it. Also, that method requires knowledge of how to setup a VM, how to properly connect USB inputs through to the VM, and how to manipulate files stored on the computer or an external hdd(ssd) to the VM. While I'm sure it works, more often than not, people won't want to go through the hassle of it.

Apaczer, 2024-03-31:
There is WIP project under: https://github.com/relalis/macos-btrfs which may provide native support for BTRFS under Mac in the future. However until then you're bound to use MTP protocol or VM method.

closed 2024-03-31 by Apaczer