TriForceX/MiyooCFW #577: [BUG]: controls are off - bittboy V2

TheMGilroy, 2024-04-11:
### Version

2.0.0 (BETAv2) pre-release

### Version type

uClibc (compatible)

### Issue description

The left button of the D pad I believe is mapped to select as it brings up the shutdown menu when I press it. Also, with no shoulder buttons, I'm unable to navigate the tabs of the UI. 

### Steps to reproduce

turn on
go through first boot procudure

### What device are you using?

BittBoy v2

### Issues' terms

- [X] I did not find any existing open/closed issue of same matter

TriForceX, 2024-04-12:
Hi, when you did the first setup which variant you selected? because there are 2 for bittboy v2 (bittboy2x_v1 and bittboy2x_v2), the issue happens with both?

TriForceX, 2024-04-12:
I see, so is not recognizing your console, you may have an hybrid-hardware, the last years many bittboy v2 were reported with that, but dont worry we can try more stuff to get it working properly, atelast screen and keyboard works.

First thing is copy and paste here the text (or file) of **uEnv.txt** from boot partition, so we can figure out some things.

Second thing to try is after first setup go to boot partition and edit console.cfg, see which value is set in CONSOLE_VARIANT, then change to bittboy2x_v1. (it should be `CONSOLE_VARIANT=bittboy2x_v1`) if doesnt work try **bittboy2x_v2**

TheMGilroy, 2024-04-12:
Here's what eUnv looks like - I tried the bittboy2x_v1 and bittboy2x_v2 and they didn't fix the issue, but v2 did make the screen flip upside down for some reason, haha. 

`CONSOLE_PARAMETERS=debug=1
CONSOLE_VARIANT=unknown
CONSOLE_VIDEO=r61520fb.ko
DETECTED_VERSION=UNKNOWN
READID_0x00=00 00 00 00
READID_0x04=00 00 00 00
arch=arm
baudrate=115200
board=sunxi
board_name=sunxi
boot_a_script=load ${devtype} ${devnum}:${distro_bootpart} ${scriptaddr} ${prefix}${script}; source ${scriptaddr}
boot_device=mmc0
boot_extlinux=sysboot ${devtype} ${devnum}:${distro_bootpart} any ${scriptaddr} ${prefix}${boot_syslinux_conf}
boot_net_usb_start=usb start
boot_prefixes=/ /boot/
boot_script_dhcp=boot.scr.uimg
boot_scripts=boot.scr.uimg boot.scr
boot_syslinux_conf=extlinux/extlinux.conf
boot_targets=fel mmc_auto usb0 
bootargs=console=tty0 console=ttyS1,115200 panic=5 rootwait quiet lpj=1015808 root=/dev/mmcblk0p2 ro
bootcmd=env export -t 0x81000000; fatwrite mmc 0:1 0x81000000 uEnv.txt ${filesize}; run distro_bootcmd;
bootcmd_args=setenv bootargs ${bootargs} miyoo_kbd.miyoo_ver=1 miyoo_kbd.miyoo_layout=1 miyoo.miyoo_snd=1 miyoo-video.driver='r61520fb+debug=1'
bootcmd_fel=if test -n ${fel_booted} && test -n ${fel_scriptaddr}; then echo '(FEL boot)'; source ${fel_scriptaddr}; fi
bootcmd_mmc0=devnum=0; run mmc_boot
bootcmd_mmc1=devnum=1; run mmc_boot
bootcmd_mmc_auto=if test ${mmc_bootdev} -eq 1; then run bootcmd_mmc1; run bootcmd_mmc0; elif test ${mmc_bootdev} -eq 0; then run bootcmd_mmc0; run bootcmd_mmc1; fi
bootcmd_usb0=devnum=0; run usb_boot
bootdelay=0
bootm_size=0x1700000
console=ttyS0,115200
cpu=arm926ejs
dfu_alt_info_ram=kernel ram 0x80500000 0x1000000;fdt ram 0x80C00000 0x100000;ramdisk ram 0x80D50000 0x4000000
distro_bootcmd=for target in ${boot_targets}; do run bootcmd_${target}; done
fdt_addr_r=0x80C00000
fdtcontroladdr=80e57890
fdtfile=suniv-f1c100s-generic.dtb
fileaddr=81000000
filesize=34
kernel_addr_r=0x80500000
mmc_boot=if mmc dev ${devnum}; then devtype=mmc; run scan_dev_for_boot_part; fi
mmc_bootdev=0
partitions=name=loader1,start=8k,size=32k,uuid=${uuid_gpt_loader1};name=loader2,size=984k,uuid=${uuid_gpt_loader2};name=esp,size=128M,bootable,uuid=${uuid_gpt_esp};name=system,size=-,uuid=${uuid_gpt_system};
preboot=usb start
pxefile_addr_r=0x80D00000
ramdisk_addr_r=0x80D50000
scan_dev_for_boot=echo Scanning ${devtype} ${devnum}:${distro_bootpart}...; for prefix in ${boot_prefixes}; do run scan_dev_for_extlinux; run scan_dev_for_scripts; done;
scan_dev_for_boot_part=part list ${devtype} ${devnum} -bootable devplist; env exists devplist || setenv devplist 1; for distro_bootpart in ${devplist}; do if fstype ${devtype} ${devnum}:${distro_bootpart} bootfstype; then run scan_dev_for_boot; fi; done; setenv devplist
scan_dev_for_extlinux=if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${boot_syslinux_conf}; then echo Found ${prefix}${boot_syslinux_conf}; run boot_extlinux; echo SCRIPT FAILED: continuing...; fi
scan_dev_for_scripts=for script in ${boot_scripts}; do if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${script}; then echo Found U-Boot script ${prefix}${script}; run boot_a_script; echo SCRIPT FAILED: continuing...; fi; done
scriptaddr=0x80C50000
silent=1
soc=sunxi
splashfile=miyoo-splash.bmp
splashimage=0x80000000
splashsource=mmc_fs
stderr=serial,vga
stdin=serial,usbkbd
stdout=serial,vga
usb_boot=usb start; if usb dev ${devnum}; then devtype=usb; run scan_dev_for_boot_part; fi
uuid_gpt_esp=c12a7328-f81f-11d2-ba4b-00a0c93ec93b
uuid_gpt_system=69dad710-2ce4-4e3c-b16c-21a1d49abed3
 `

TriForceX, 2024-04-12:
I will take a look on it later, thanks!

btw when you said "but v2 did make the screen flip upside down for some reason, haha." did the controls worked fine? you can re-flip again by changing FLIP=0 to 1 in options.cfg file inside main partition.

closed 2025-09-03 by Apaczer