TriForceX/MiyooCFW #590: usb host at the startup

CountessTemmie, 2024-05-19:
would be nice an option to turn on the usb host at the startup for those that play with controllers a lot

Apaczer, 2024-06-22:
The future idea is to display a prompt in gmenu2x (if USB is connected) to change USB options at use: 
```
< > Charge
<x> File Transfer
< > Connect Device
< > Serial Port.
```

 Same goes for MiniJack connection prompt (if detected) to change Data being redirected:
 ```
 <x> Audio output (headphone
 < > Audio/Video output (TV)
 ```

Apaczer, 2024-06-22:
For now u could edit file `/etc/main` from rootfs in this part:
https://github.com/MiyooCFW/buildroot/blob/fd34e2b21ecd39476599dfcee493048c1f02986e/board/miyoo/rootfs/etc/main#L130
by adding cmd from `usb-host.sh` script app, just before gmenu2x startup

Apaczer, 2025-11-09:
same goes for TV/Audio jack input via: https://github.com/MiyooCFW/gmenu2x/pull/99

However no auto-detection on the PocketGO cuz it have audio jack + extra tv jack port and for latter we have no info about specific pin reading