TriForceX/MiyooCFW #591: [BUG]: Retroarch Menu Swap OK and Cancel Buttons broken

bishopdynamics, 2024-05-25:
### Version

2.0.0 (BETAv2) pre-release

### Version type

uClibc (compatible)

### Version hash

67298a0

### Issue description

In Retroarch -> Main Menu -> Settings -> Input -> Menu Controls, the option "Menu Swap OK and Cancel Buttons" has unexpected behavior: 
When disabled, A=OK, B=Cancel (correct behavior)
When enabled, A=OK, B=OK (incorrect behavior), expected: A=Cancel, B=OK

### Steps to reproduce

1. Turn on the device
2. Launch RetroArch
3. Observe default menu navigation behavior: A=OK/Enter, B=Cancel/Back
4. Main Menu -> Settings -> Input -> Menu Controls
5. Enable the option "Menu Swap OK and Cancel Buttons"
6. Observe that A and B both behave as OK

### What device are you using?

PowKiddy Q90

### Issues' terms

- [X] I did not find any existing open/closed issue of same matter

bishopdynamics, 2024-05-25:
For anyone experiencing the same issue, I found a workaround that might actually be "the right way" to do this anyway.
In GMenu2x -> Settings, try setting "Keyboard Layout" to 2. 
That layout seems to swap A and B for me on Q90/V90, and seems to be a more universal method.

Apaczer, 2025-04-30:
I believe we should be using "Unified Menu controls" so that RA treats keystrokes the same way as retropad so there is no confussion when using menu (mostly notorious in "on screen keyboard"). Will be closing after: https://github.com/MiyooCFW/buildroot/pull/166

Apaczer, 2025-05-12:
closing via: https://github.com/MiyooCFW/buildroot/commit/8037d1c1e2fffc3a7dd1d8ab790158d4f9ec8019

closed 2025-05-12 by Apaczer