TriForceX/MiyooCFW #603: [BUG]: Add video filter freeze the display

dem1980, 2024-07-15:
### Version

2.0.0 (BETAv2) pre-release

### Version type

musl (experimental)

### Issue description

When you add/activate a video filter or shader in retroarch, and go back in game, the display seems to be broken. I have black screen.
If you quit the game/retroarch and start a new game, the video filter seems to be well activated.

### Steps to reproduce

1. Start a game with retroarch core
2. Activate a shader
3. Go back to game, the screen remains Black
4. Quit the game, push hotkey button, press B (without view the screen, it is broken), move Up, press A, the game will be closed and you are back in MiyooCfw
5. Start a game, the shader is activated and works

### What device are you using?

PowKiddy V90

### Issues' terms

- [X] I did not find any existing open/closed issue of same matter

Apaczer, 2024-07-23:
Correct, you have to enbale Vsync in general settings to bypass this bug.
Unfortunately this has been an issue since day one (when the RetroArch have been ported officially to Miyoo). 

We disabled Vsync by default because there are artifacts on pcsx_rearmed core when it is active (probably due to DOUBLEBUFFERING being active with it). Moreover you have to disable core_specific hotkey FF input (and enable global) if you want to use fast forward at all in that configuration.

Probably we will enable Vsync in the future for good with proper settings.

dem1980, 2024-08-19:
Thank you.
The parameter Vsync resolve this problem when i need to test/setup shaders.

Apaczer, 2026-05-14:
fixed in upstream and soon in our fork, I've added also "Video filter enable" hotkey option if you would want to test various filters on the fly while running content without menu.