TriForceX/MiyooCFW #630: [FEATURE-REQ]: alternative fast-fs partition for CD-based games

jelazual, 2025-04-01:
## 💡 Feature request

High performance/light weight filesystem for reading generic game data, safe/stable filesystem for critical system data.

## Motivation (behind your idea)

cd-based games encounter big lags whenever reading from the SD card slot, at least on Powkiddy V90. CD Audio periodically stutters in Turbo CD game, for instance, or when loading or saving.
I can't find where the mounts are defined(root/etc/fstab doesn't have them?) so I can't redefine partitions and remount things, such as have btrfs MAIN and a large xfs EXTEND partition for CD games.

BTRFS is a great filesystem, but it expects a lot of CPU to perform well, so it's not great for loading ROMs or maybe even saving ROM saves or save states, less importantly.
It should probably be fine to back up ROM saves and save states when loading a ROM, in case of filesystem partial writes causing data loss, I think XFS or EXT4 can recover from partial writes with only some data loss to what is being written. Doesn't it even have a reflink/redirect on write mode, maybe that can be used? But that might be a lot more work.

It would also save some amount of battery overall for CD-based games to use a lighter FS like XFS or EXT4, since fewer CPU cycles are run to load data such as music in the background.

Currently, to do this seems to require deleting MAIN, turning it into an extended partition, and then making a new BTRFS MAIN and XFS/EXT4 extra partition, and then hijacking the current boot/mount operations to map the locations appropriately. But, I don't know where this mount and mapping is stored and have not successfully tried this so far.

nfriedly, 2025-04-01:
Yeah... I was the one who originally pushed for BTRFS, and I still think it's a great filesystem, especially for the Linux partition, but I am seeing some of the drawbacks of using it for ROMs.

Maybe it would be wise to offer a choice of filesystem for the ROMs partition during the initial setup: 
* BTRFS for reliability 
* FAT32 for compatibility 
* Perhaps EXT4 or, if we can make it work, NTFS for an in-between option

Maybe also offer to put saves on a separate BTRFS partition to make them more reliable?

Apaczer, 2025-04-02:
Haven't tried PCE games in a while, but revisited CD _Rondo of Blood_ now and must say loading state is instantous (saving takes 1s more). Moreover no or little hick-ups during gameplay for this game, the same can be told about Mega-CD with picodrive. 
What specific ROM is giving you a hard time? Are you using included Beetle PCE FAST libretro core? Did you try this game on FAT fs with the same core?
We could add CHD support for this specific core (there is a PR pending for that to happen).

jelazual, 2025-04-03:
I don't think it's an issue with data rate, so I don't think adding compression would necessarily benefit here, outside of reducing the number of bytes to load through btrfs.
In Tengai Makyou Ziria, there's frequent stutters during the CD audio in the opening, especially at the start when it's preloading the title credits and title screen data, whenever a character is talking during cutscenes, when loading new maps/changing screens, when appearing monsters at the start of combat, when summoning exp and gold quantities at the end of combat, when summoning text of NPCs, and when saving or loading the game.
I have not tried other filesystems because I cannot add an additional partition or configure the mounts via fstab with the 2.0.0 beta btrfs powkiddy v90 image. However, the stutters are not sudden stalls while waiting for data to show up, but stutters as if heavy CPU util is going on when loading **or saving** data.
I am using the included PCE libretro core.
The game does appear to be on the demanding side, as underclocking to 600mhz or so makes the game run slightly slow, but it otherwise runs full speed at 720mhz comfortably, until reading or writing any data.

I forget what else I tried that had similar issues, maybe something Sega CD but at the time I wrote it off as the emulator being too marginal as performance in general wasn't top either. But, for turbo CD, Ziria runs very very smooth in all other cases than reading or writing any amount of data.

Apaczer, 2025-04-07:
I've tried _Tengai Makyou: Ziria_ and that game is just too demanding for Mednafen's PC Engine lr-core implementation. It doesn't matter if you launch from FAT partition (same kernel etc. but no difference in perf.), or you launch game from pre-cached instance (did a CHD convert but using `zstd` in custom libretro build for fast decompression). Lucky for you smbd (that is Exophase) out of bordom wrote an PC-Engine emulator called `temper` with heavy ARM assembly optimization, which makes every game runs 3x as fast as it would be normaly on libretro `beetle-pce-fast` - so there is no hiccups mentioned by you on [standalone](https://github.com/Apaczer/temper/releases/download/1.25.m1/temper.ipk) version of this emu.

EtherealIntellect, 2025-04-09:
I'd like to add ps1 on the list of emulators strugging when loading files - its not really talked about since ps1 barely runs anyway, but trying megaman x4 and tony hawk 3 very obviously lag extra hard when new stuff needs to load into view (which wasn't as noticeable on the super old ones before btrfs). There is also another post mentioning in discussions https://github.com/TriForceX/MiyooCFW/discussions/629

Ive only tried standalone so far, not the latest latest retroarch from this month but idk how different it is

eku, 2025-04-09:
> BTRFS is a great filesystem, but it expects a lot of CPU to perform well, so it's not great for loading ROMs or maybe even saving ROM saves or save states, less importantly.

Do you have a source with reliable numbers or is this just your gut feeling?

jelazual, 2025-04-09:
> > BTRFS is a great filesystem, but it expects a lot of CPU to perform well, so it's not great for loading ROMs or maybe even saving ROM saves or save states, less importantly.
> 
> Do you have a source with reliable numbers or is this just your gut feeling?

Yes. I use btrfs! For heavy file I/O operations on fast drives, it very quickly becomes CPU limited. For random I/O performance, on a fast enough SSD, even high end CPUs will be CPU limited in BTRFS over EXT4/XFS or similar.
It may not be responsible for the lags here, but it absolutely is a very feature-rich, CPU heavy, and relatively slow filesystem, like ZFS is. 

I don't think FAT32 is particularly fast either, though; at least in my experience, xfs has been the best performer across the board, but I've seen some people report better results with ext4 instead.

closed 2025-05-18 by Apaczer