TriForceX/MiyooCFW #641: [BUG]: Error when Expanding Partition

M4d3v1l5, 2025-05-24:
### Version

2.0.0 (BETA) pre-release, 2.0.0 (BETAv2) pre-release

### Version type

musl (experimental), uClibc (compatible)

### Issue description

Welp, i got an error message when trying to expand the partition on all cfw 2.0.0 version. 

[image]
[image]

its fine and continued the 1st booting sequence if i chose "NO" option, but it got stuck at those screen if i choose "YES". Is there any alternative option or ways to manually expand those Main partition? as its now is not even accessible in cfw 2.0.0. Without an access to most of the SD card space, i can't do nothing much sadly. Guess ill be back to 1.3.3 again.

The new usb transfer format where you didn't have to eject the SD is good in theory, but in the implementation its become more of a hasle when a problem arise, as now we lost the access to Main partition directly without needed extra tools.

Will there be an alternative build with the older storage format (like 1.3.3) as alternative? or this will be the now standard of version 2 forward?

### Steps to reproduce

normal installation like the usual

### What device are you using?

Sup M3 v1, PowKiddy V90

### Relevant log outputs

```shell

```

### Issues' terms

- [x] I did not find any existing open/closed issue of same matter

Apaczer, 2025-05-28:
Did you try with another SD card (different size and brand)? The `input/ouput error` may indicate it's a HW issue.

M4d3v1l5, 2025-06-05:
i am trying that on 2 diferent sd card : Sandisk & V-gen, both stuck at that same step. Trying to expand will just cause looping to 1 instalation sequence (where you choose screen type) on V90 and straight up crashing and corrupting the image on M3.

Strange thing is : reinstalling old CFW pre new storage format beta 2.0~1.3.3 work fine on both sd card, so it shouldn't be hardware problem no?