TriForceX/MiyooCFW #643: [BUG]: can't change brightness settings on XYC Q8

less-ly, 2025-07-18:
### Version

2.0.0 (Dev_build) nigthly

### Version type

uClibc (compatible)

### Issue description

There seems to be no way to change brightness on xyc q8 with either console hotkeys or via gmenu2x settings menu.

### Steps to reproduce

1. Turn the device ON.
2. a. Press either "VOL + →" or "VOL + ←".
2. b. Or go to the settings menu, change brightness value and save settings. 
3. Thus, the issue occurs.

### What device are you using?

XYC Q8

### Relevant log outputs

```shell

```

### Issues' terms

- [x] I did not find any existing open/closed issue of same matter

Apaczer, 2025-09-03:
Could you try to replace this [suniv-f1c100s-miyoo.dtb](https://github.com/user-attachments/files/22125907/suniv-f1c100s-miyoo.zip) file (unzip & backup the old-one) in your BOOT partition and see if there is improvement? This DTB file is for CFW you've been using or older, not the latest one.

We believe the [brightness levels](https://github.com/MiyooCFW/kernel/blob/c2f37dbd03d583a6ce5c83cd94dd46f88d5ac91f/arch/arm/boot/dts/suniv-f1c100s-miyoo.dts#L26) should be lower on your XYC device.

less-ly, 2025-09-07:
Just to be on a saner side,
the Version 2.0.0 Beta (Pre-Release)* 
had been flashed on the XYC's microSD,
the system was initialized,
and the [suniv-f1c100s-miyoo.dtb](https://github.com/user-attachments/files/22125907/suniv-f1c100s-miyoo.zip) file was unziped
and placed instead of the existing one in the BOOT partition.
 
Sadly, there's no improvement.

\* \- brigtness levels can't be changed on neither beta pre-release, nor nightly builds

Apaczer, 2025-12-26:
kernel creates uint32-array table from steps provided in brightness-levels (see in src: Documentation/devicetree/bindings/leds/backlight/pwm-backlight.yaml). I was told that removing ~~leading~~ trailing 0 fixed the issue in another XYC device. Since this is rather range problem, try with this DTS: 
[suniv-f1c100s-miyoo.zip](https://github.com/user-attachments/files/24350612/suniv-f1c100s-miyoo.zip)
(_sha256 4039652f797fdf9ca3fc556bc7318426b5ef5bd838d093a6db655eb8551a3653_)

NOTE: this is for latest nightly build, not sure if this will work on older

less-ly, 2025-12-27:
The spiked brightness-levels

> brightness-levels = <0x00 0x01 0x03 0x05 0x08 0x0a 0x0c 0x0f 0x12 0x14 0x64>

in the [suniv-f1c100s-miyoo.zip](https://github.com/user-attachments/files/24350612/suniv-f1c100s-miyoo.zip)
(_sha256 4039652f797fdf9ca3fc556bc7318426b5ef5bd838d093a6db655eb8551a3653_)
introduce an interesting behavior to the mix:
```
10-40   will turn the screen completely off
50-70   will uniformly give a less bright screen
80-100 are uniformly about a more bright screen
```

Checked on the latest nightly build.